carla-simulator/scenario_runner #457: Support for UserDefined actions in OpenSCENARIO

amstrudy, 2020-02-14:
Adding support for the `UserDefined` action in OpenSCENARIO would allow for a greater degree of flexibility when crafting scenarios. This is not currently supported.

I think the "single" and "continuous" execution types could correspond to the Action either completing after the python script completes, or as soon as the python script is started (script continues executing indefinitely).

This example would run the filename.py script, and once that completes it would signal the end of this Action:
```xml
<UserDefined>
     <Script name="startScript" execution="single" file="filename.py" />
</UserDefined>
```

amstrudy, 2020-02-14:
Wanted to make a formal feature request for this to make sure that there wasn't something in the works before I begin working on it.

fabianoboril, 2020-02-14:
I just added something initial and preliminary yesterday. I opened a PR, but had no time yet to completely test and cleanup the code (see #458).
Currently the filename has to include the arguments, e.g. "/path/to/file.py arg1 args2".

amstrudy, 2020-02-14:
Ha ok! In the time between me opening this and you responding I came up with pretty much line by line the same thing as you. I will use what I have until yours is accepted.

PR #458 (2020-02-24): Added initial support for OSCUserDefinedActions with script exec
Fixes #457

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/458)
closed 2020-02-24 by fpasch