carla-simulator/scenario_runner #459: OpenSCENARIO Catalog support

amstrudy, 2020-02-17:
Has there been any work to support catalogs in OpenSCENARIO? They would be very useful when making a bunch of similar scenarios.

A use case would be when creating a new entity one could simply having a catalog reference rather than re-defining the same vehicle in multiple files.

I don't see any other requests for it, but I wanted to check before beginning to implement a solution.

fabianoboril, 2020-02-20:
Hi,
we did not yet start working on this. If you have anything working we would be very happy if you could share it with us and the community.

amstrudy, 2020-02-21:
@fabianoboril Just a quick question: what is the purpose of lines 258-260 of `open_scenario.py` shown here
```python
parallel_actions.add_child(
    oneshot_behavior(variable_name=get_xml_path(self.config.story, child),
                    behaviour=maneuver_behavior))
```
Due to the nature of catalogs, the `get_xml_path()` is proving to be difficult. What is that path used for?

PR #465 (2020-03-02): Initial support for OpenSCENARIO Catalogs

#### Description

Fixes #459 

These changes allow for the use of the Vehicle, Pedestrian, Environment, MiscObject, and Maneuver OpenSCENARIO Catalogs. By including a catalog file, one can simply reference one of its entries in multiple scenarios. 

This is done by reading in all of the catalog entries into a dictionary that resides in `OpenScenarioConfig`. In the places where Vehicles/Pedestrians/MiscObjects/Environments are read from the scenario file, the catalogs are simply used instead. Maneuvers were a bit trickier but again if a `CatalogReferen [... cut]
closed 2020-03-02 by fabianoboril