carla-simulator/scenario_runner #464: OpenSCENARIO vehicle not following route

amstrudy, 2020-02-25:
I am looking to have my vehicle (ego or otherwise) to follow a set of waypoints, which should be supported through the <Route> tag and the waypoint following atomic behavior. For the life of me I cannot get the vehicle to start following the points however. It works fine with a pedestrian, but neither the ego nor another vehicle will drive unless I just turn on the autopilot.

With this specification, I expect the vehicle to rather erratically drive to try and reach these points but nothing happens:
```xml
<Sequence name="RoutingSequence" numberOfExecutions="1">
    <Actors>
        <Entity name="vehicle" />
    </Actors>
    <Maneuver name="FollowingRouteManeuver">
        <Event name="FollowsRoute" priority="overwrite">
            <Action name="FollowsRoute">
                <Private>
                    <Routing>
                        <FollowRoute>
                            <Route name="StraightRoute" closed="false">
                                <ParameterDeclaration />
                                <Waypoint strategy="fastest">
                                    <Position>
                                        <World x="-7.691344" y="-18.206945" z="4" h="-0.4" />
                                    </Position>
                                </Waypoint>
                                <Waypoint strategy="fastest">
                                    <Position>
                                        <World x="13" y="-20" z="0" />
                                    </Position>
                                </Waypoint>
                                <Waypoint strategy="fastest">
                                    <Position>
                                        <World x="15" y="-30" z="0" />
                                    </Position>
                                </Waypoint>
                            </Route>
                        </FollowRoute>
                    </Routing>
                </Private>
            </Action>
            <StartConditions>
                <ConditionGroup>
                    <Condition name="StartCondition" delay="0" edge="rising">
                        <ByValue>
                            <SimulationTime value="0" rule="greater_than" />
                        </ByValue>
                    </Condition>
                </ConditionGroup>
            </StartConditions>
        </Event>
    </Maneuver>
</Sequence>
```

I couldn't find any examples that use the Route follower. What am I missing here?

Ubuntu 18.04, Carla 0.9.6

fabianoboril, 2020-02-26:
Hi,
have you tried to use more than 5 waypoints? I think there was a restriction from CARLA's planning module that requires more than 5 waypoints.

amstrudy, 2020-02-26:
Yes! That was the issue. Thank you.

closed 2020-02-26 by amstrudy