carla-simulator/scenario_runner #471: Update to 0.9.7 "cloudyness" bug

amstrudy, 2020-02-28:
**Describe the bug**
There's a really small typo on line 361 in `scenario_runner.py` where it says cloudyness but should be cloudiness to comply with the CARLA 0.9.7 PythonAPI.

**To Reproduce**
Steps to reproduce the behavior:
1. I just ran a simple OpenSCENARIO example: `python3 scenario_runner.py --openscenario srunner/examples/CyclistCrossing.xosc` and got this error:

```
Traceback (most recent call last):
  File "scenario_runner.py", line 620, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 610, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 526, in run
    result = self._run_openscenario()
  File "scenario_runner.py", line 516, in _run_openscenario
    result = self._load_and_run_scenario(config)
  File "scenario_runner.py", line 366, in _load_and_run_scenario
    sun_altitude_angle=config.weather.sun_altitude
Boost.Python.ArgumentError: Python argument types in
    WeatherParameters.__init__(WeatherParameters)
did not match C++ signature:
    __init__(_object*, float cloudiness=0.0, float precipitation=0.0, float precipitation_deposits=0.0, float wind_intensity=0.0, float sun_azimuth_angle=0.0, float sun_altitude_angle=0.0, float fog_density=0.0, float fog_distance=0.0, float wetness=0.0)
    __init__(_object*)
```
Line 361 should just be `cloudiness=config.weather.cloudyness,` instead.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version 0.9.7
 - Python version 3.6.9
 - Version 0.9.7

I figured this didn't merit a whole pull request, so just submitting as a bug instead. This issue started for me as soon as I switched to 0.9.7.

fabianoboril, 2020-02-28:
This was mentioned also in #445. The problem here is that CARLA has a 0.9.7 release, and patch versions that came after. This is also highlighted in our README. We do not yet support the patch versions. Sorry for this trouble. We will have a fix for this soon (in master).

amstrudy, 2020-02-28:
@fabianoboril Ok, thanks for the clarification. 

Aside from the "cloudiness" issue, I still cannot successfully set the weather through OpenSCENARIO. For example, in the `PedestrianCrossingfront.xosc` example, the weather is set to clear and dry but the world is rainy.

fabianoboril, 2020-02-29:
PR #474 Addresses the weather issues. Adaptations are again possible.

PR #474 (2020-03-02): Fixed weather with CARLA 0.9.7.x
- Adapted to API change (cloudyness -> cloudiness)
- Updated OpenSCENARIO weather parser

Fixes #471, #445

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/474)
closed 2020-03-02 by fabianoboril