carla-simulator/scenario_runner #480: Hi, I'm working on my master thesis and I'm using Carla to implement Reinforcement learning algorithm. I want to test my agents on a particular scenario, so I'm using Carla scenario_runner. I have a problem: while the scenario runner is waiting for ego, even if I spawn my vehicle it doesn't interact with the scenario. But, meanwhile my agent is going around in the town if i use manual_control.py the new vehicle is spawning on the correct point and triggers the scenario runner, so the scenario starts. From the manual_control.py I copied the way to spawn the vehicle, still doesn't work. Can you please help me?

giovox, 2020-03-04:
Hi, I'm working on my master thesis and I'm using Carla to implement Reinforcement learning algorithms.
I want to test my agents on a particular scenario (DynamicObjectCrossing), so I'm using Carla scenario_runner. The idea is to do the same scenario for a set of episodes and let the agent learn to brake in that situation.

I have a problem: I can't handle repetitions, because every time a new episode starts the scenario and my agent lose synchronise. So sensors will go out of scope or my agent try to kill something already dead. Can you please help me?

 - OS: [Ubuntu 18.04]
 - CARLA Version [CARLA 0.9.7.4]
 - Python version [3.5]
 - Version [0.9.7]

fabianoboril, 2020-03-04:
Hi,
can you explain (maybe with a short example on commandline) what exactly you try to do?

giovox, 2020-03-04:
I am running on a terminal: 
python scenario_runner.py --scenario DynamicObjectCrossing_1 --reloadWorld --repetition 3 --waitForEgo

while in another terminal I run a modified version of manual_control.py where I added 
   - on the World class
        def step(self, action):
             if action == 0:
                 self.player.apply_control(carla.VehicleControl(throttle=1.0, steer=-1))
             elif action == 1:
                 self.player.apply_control(carla.VehicleControl(throttle=1.0, steer= 0))
             elif action == 2:
                 self.player.apply_control(carla.VehicleControl(throttle=1.0, steer=1))
   - I changed the game_loop() in this way
        try:
          start = time.time()
          timeout = 15
          client = carla.Client(args.host, args.port)
          client.set_timeout(2.0)
          for i in range(1, 4):
              hud = HUD(args.width, args.height)
              world = World(client.get_world(), hud, args)
              while (time.time()-start) <= timeout:
                  action = 1
                  control = world.step(action)
                  time.sleep(1/60)
              world.destroy()
              world.destroy_sensors()
      finally:
          if world is not None:
              pass
        pygame.quit()

the first repetition the car moves and iteract with the scenario, the second repetition the car spawn but it stays idle and I have a warning on the second terminal WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1608 (sensor.camera.rgb) so the manual_control.py crashes.

I am a beginner in programming so I apologize in advance for my errors.

glopezdiest, 2021-11-23:
Closing this issue while waiting for the original poster to provide more information. Feel free to reopen it if needed

closed 2021-11-23 by glopezdiest

BadrBenElallid, 2022-03-04:
Hi 
 
Please, can you explain to me how to test agents on a particular scenario in the Carla simulator?