carla-simulator/scenario_runner #484: How can I observe the scenario simulation for version 0.9.5 for a simple scenario

xgnit, 2020-03-10:
OS: windows 10
Carla version: 0.9.5
Python Version 3.7
Version: 0.9.5.1

I was trying to run a simple scenario by start the scenario runner,
1. I started the Carla on windows 
2. I started scenario runner by: 
    python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

Then the Carla server window switched to the new scene, but no vehicles was spawned. If I do nothing, the scenario runner shows:

Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Failure!
ERROR: failed to destroy actor 267 : unable to destroy actor: not found

Is it supposed to be like this?

3. Then I tried to start the manual control with autopilot mode by python manual_control.py -a, then I can see the ego and leading vehicle were both spawned.

But I thought this manual control is only for the "manual control"?

xgnit, 2020-03-10:
I read the source code:
it will wait for the user controlled vehicle to
        enter the start region, then make the other actor to drive until reaching
        the next intersection. Finally, the user-controlled vehicle has to be close
        enough to the other actor to end the scenario.

Seems this scenario is supposed to wait for any client to create a ego vehicle actor and the leading vehicle.

xgnit, 2020-03-10:
I am not sure about my guess, can someone confirm it for me? thnx

fabianoboril, 2020-03-10:
Hi,
you have to start the manual_control.py to see the vehicles. The problem is that no CARLA client is by default connected to the ego vehicle to let it move. Once you have such a client (e.g. the manual_control.py) and start driving, things will "magically" start to happen.

xgnit, 2020-03-10:
@fabianoboril  Thnx!

closed 2020-03-10 by xgnit