carla-simulator/scenario_runner #493: Argument Error with running route-based scenario

aroongta, 2020-03-22:
**Argument Error with running route-based scenario**
The getting-started documentation for the scenario runner has an example to use the route-based scenario. The "route" argument seems not a valid argument for the scenario_runner.py script. 
`python scenario_runner.py --route <path/to/route-file> <path/to/scenario_sample_file> [route id]`

Therefore I receive an error saying the argument is unrecognized. 

**Expected behavior**
I expected it to launch the Carla AD challenge route. 

**Screenshots**
[image]

I am trying to generate a route based test track to have users drive using a Logitech G920 controller and record cognitive signals. Please look into the problem and let me know the best way to debug it. 

**Desktop and Carla Specs:**
 - OS: Ubuntu 16.04
 - CARLA Version [CARLA 0.9.6]
 - Python version [ 3.5]
 - Version [ 0.9.6]

fabianoboril, 2020-03-24:
Try to add --route in front of your arguments.

aroongta, 2020-03-26:
Hi @fabianoboril ,
I did try to add --route, still get the same unrecognized argument error. 
Also, I cannot find the --route argument definition in the scenario_runner.py. I am not sure if it is supposed to be like that. 
Please look into it and let me know.

Thank you

glopezdiest, 2020-04-09:
Hello @aroongta, what version of scenario_runner are you using? The --route argument has been there for a very long time now and you should most definitely have it.

aroongta, 2020-04-09:
Hello @glopezdiest ,
I am using the 0.96. 
Figured it, the zipped files for some reason are lacking the argument definition. I cloned the repository, those seem to have all the definitions. Probably the zipped folders are not updated. 

Thank you.

closed 2020-04-09 by aroongta

gurpreet2-kaur, 2022-02-09:
The example scenario file all_towns_traffic_scenarios.json given in Carla code doesnot work for route based scenario fails with error " Some scenarios are failing"
Any reference  to a simple scenario configuration file so that I can successfully run a route based scenario.