carla-simulator/scenario_runner #494: Not able to run scenario

AMRITKV88, 2020-03-24:
When I am trying to run the follow vehicle scenario using the command :

`python3 scenario_runner.py --reloadWorld  --scenario FollowLeadingVehicle_1`

and manual_control.py on other terminal using the command : 

 `python3 manual_control.py --autopilot`

The pygame window pops-up and it runs, but still I am getting such **Failure** errors as mentioned below.

I am getting following errors in scenario terminal :

```
Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Resetting ego-vehicle!
Destroying ego-vehicle!
Failure!
Destroying ego vehicle 434
No more scenarios .... Exiting
```

And in the manual_control.py terminal the error is:

```
Scenario ended -- Terminating
ERROR: failed to destroy actor 434 : unable to destroy actor: not found 
Fatal Python error: PyEval_SaveThread: NULL tstate
```
I am quite new to this CARLA, so I followed the getting_started tutorial. But not able to handle this error. Please help me out.

**Screenshots**
Here I am attaching one such failure run screenshot.
[image]

**Desktop:**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.8]
 - Python version [e.g. 3.6]
 - Version [e.g. 0.9.8]

**Thanks in advance....!**

fabianoboril, 2020-03-24:
Hi,
the scenario was successfully executed, but some of the evaluation conditions (i.e. tests) failed. Please run with --output to get further info.

AMRITKV88, 2020-03-26:
@fabianoboril Thanks for the update. I tried the same, but I am unable to understand completely. As per my understanding its showing that expected time is 60.00s but it took 60.05s. Am I right ? But again in system time its showing 17.22s.  Is there anything I am missing or something other which need to be changed. Please suggest the edits required.
I am attaching a screenshot here for your reference; after running the command with --output keyword. Please help.

[image]

fabianoboril, 2020-03-30:
The CARLA time can be faster than system time depending on the performance of your system, rendering quality, amount of sensor etc. Please see the CARLA documentation for further information.

villinux, 2020-04-06:
Hi 
I have the same problem.
Every scenario i try to run failes after 10 to 20 seconds. I have already tried --output but i get same result as mentioned above. How can this be fixed?

fabianoboril, 2020-04-06:
Well, this depends on the exact failure. Did you connect an agent (e.g. by running manual_control.py) to the ego vehicle, and drove around? If this is the case, take a look at the exact condition that failed. All scenarios are designed to just run for a fixed amount of (CARLA) time, and afterwards terminate.

villinux, 2020-04-06:
I tried manual_control but i can only drive while the scenario is running. When it fails the manual control terminates immediatly. 
I haven't used an agent yet

closed 2020-04-16 by fabianoboril

zeinebghaddab, 2022-08-09:
Hi , i'm trying to simulate ego vehicles on carla using openscenario file, i don't want to control the vehicle using manual control, i want that  the vehicle  follow a specific path that i define. could anyone help me with openscenario example?

TigerStone93, 2022-11-01:
> The CARLA time can be faster than system time depending on the performance of your system, rendering quality, amount of sensor etc. Please see the CARLA documentation for further information.

What does it exactly mean?

PR #509 (2020-04-16): Fix to scenarios running faster than real time
#### Description

`setting.fixed_delta_seconds = 1.0 / self.frame_rate` has been moved inside the agent check, as it was being set even in asynchronous mode, causing the simulation to run faster than in real time.

Fixes #494

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.24
  * **CARLA version:** 0.9.8

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/509)