carla-simulator/scenario_runner #495: Autopilot not moving in version 0.9.8 (due to Traffic Manager?)

rueeggmi, 2020-03-24:
When using the autopilot with `vehicle.set_autopilot(True)` the vehicle doesn't start to move. I have a rather complicated set up that can be described as follows:
I'm using ScenarioRunner with a client, similar to manual control. My client has to run in synchronous mode. Because it is not possible to synchronize two clients with the CARLA server, I run my script as a thread in the ScenarioRunner, which takes as input the client started in the ScenarioRunner script. Within this threaded script, the autopilot is set to true for the ego_vehicle. This worked very nicely in the 0.9.6 version. Now I updated to version 0.9.8 and although the autopilot is set successfully, the car does not start to move. 

I checked the feasibility of this approach in the 0.9.8 version with the spawn_npc script. This runs without the ScenarioRunner. I have implemented the spawn_npc again as a thread, that is called from a client similar to the normal manual_control client (running in synchronous mode). For this setup, the spawned actors do start moving when setting the autopilot. In spawn_npc, the autopilot is set with: `batch.append(SpawnActor(blueprint, transform).then(SetAutopilot(FutureActor, True)))`. Because I didn't find any documentation online on the functions SpawnActor, SetAutopilot and especially FutureActor, I'm not sure on how to implement it like this in my own script. Furthermore, since I use the ScenarioRunner, my vehicle is spawned in the ScenarioRunner script, so I can't use the above mentioned command in the threaded script as I only want to set the autopilot but not spawn the vehicle there.

I feel like this issue is due to the new Traffic Manager. In the documentation, it is written, that in the ScenarioRunner, the TM will be subordinated to the ScenarioRunner and wait for it. Where is this implemented in the ScenarioRunner code?

Has anyone run into a similar issue or knows a possible solution for this?

Running on:
Ubuntu 18.04
CARLA Version 0.9.8
 Python version 3.6
 ScenarioRunner 0.9.8

fabianoboril, 2020-03-24:
Hi,
I just had a quick test with the manual_control.py example, and the autopilot for the ego vehicle is working just fine. Can you try to create a minimum example such that we can reproduce the problem?

rueeggmi, 2020-03-24:
If you copy these two files into the scenario runner folder, change the carla root path in the beginning of both files accordingly  and run it with: `python scenario_runner_with_thread.py --reloadWorld --preview --autopilot --scenario OppositeVehicleRunningRedLight011` you should be able to see the issue.

The thread is created in line 154 and started in line 557 of scenario_runner_with_thread.py. And the autopilot is set in line 263 of example_thread.py

[Example_Code.zip](https://github.com/carla-simulator/scenario_runner/files/4375550/Example_Code.zip)

jackbart94, 2020-04-07:
``And the autopilot is set in line 263 of example_thread.py``

You are using the SetAutopilot command that is meant for spawning vehicles in batch, without applying said command.
Down here are the correct lines of code so that it works, let me know if this solves this issue!

```
        batch = []
        # Set Autopilot
        if args.autopilot:
            #vehicle.apply_control(carla.VehicleControl(throttle=0.1, brake=0.1))
            batch.append(SetAutopilot(vehicle, True))
            client.apply_batch_sync(batch)
```

closed 2020-04-09 by glopezdiest

kargarisaac, 2020-10-11:
> `And the autopilot is set in line 263 of example_thread.py`
>
> You are using the SetAutopilot command that is meant for spawning vehicles in batch, without applying said command.
> Down here are the [...]

I have the same problem. can you take a look at https://github.com/carla-simulator/carla/issues/3424

SaeedBASALEH, 2023-05-25:
Hi. I am using the version 9.11 and I spawn three different vehicles in autopilot mode using set_auto_pilot(True) but always I have only the first one is moving, the others are in freezing mode!!

The code of spawing and set the autopilot. For information, the spawing points coordinates are not the waypoints generated directly from the map but still on the road, that means when I spawn the vehicles they are on the road !!

THE CODE:

    vw_bp = world.get_blueprint_library().find('vehicle.dodge_charger.police')
    vw_bp.set_attribute('color', '0,0,0')
    vw_spawn_point = carla.Transform(carla.Location(x = left[35][0], y = left[35][1], z = 1), carla.Rotation(yaw = -270, roll = 0.0))
    vw = world.spawn_actor(vw_bp, vw_spawn_point)
    vw.set_autopilot(True)

    toy_bp = world.get_blueprint_library().find('vehicle.toyota.prius')
    toy_bp.set_attribute('color', '255,0,0')
    toy_spawn_point = carla.Transform(carla.Location(x = right[100][0], y = right[100][1] , z = 1), carla.Rotation(yaw = -90, roll = 0.0))
    toy = world.spawn_actor(toy_bp, toy_spawn_point)
    toy.set_autopilot(True)

    kn_bp = world.get_blueprint_library().find('vehicle.kawasaki.ninja')
    kn_bp.set_attribute('color', '255,0,0')
    kn_spawn_point = carla.Transform(carla.Location(x = right[60][0], y = right[60][1], z = 1), carla.Rotation(yaw = -180, roll = 0.0))
    kn = world.spawn_actor(kn_bp, kn_spawn_point)
    kn.set_autopilot(True)

omkart, 2024-10-07:
I am using Carla 0.9.15 and I have similar issue
I use a basic scenario runner example such as the CyclistCrossing.xosc and run this using the scenario runner
I then run my own script instead of the manual_control.py. In this I am updating the ticks of the simulation manually and have kept both the world and the TM in synchronous mode. Then I find the ego vehicle spawned by the scenario. But even after setting the autopilot as below, the vehicle does not move. 

Also I see one more issue where if I change the external_control 

 here is my code

#Connect to the CARLA server running on the localhost
            client = carla.Client('localhost', 2000) # replace with 'localhost' for Linux
            #world = client.load_world('Town01')
            
            world = client.get_world()

            # Set sim parameters and running in synchronous mode
            settings = world.get_settings()
            settings.synchronous_mode = False
            settings.fixed_delta_seconds = self.delta_milliseconds / 1000.0
            # settings.quality_level = 'Low'
            # settings.no_rendering_mode = True
            world.apply_settings(settings)

            tm = client.get_trafficmanager()
            tm.set_synchronous_mode(True)

           possible_vehicles = world.get_actors().filter('vehicle.*')
            for vehicle in possible_vehicles:
                if vehicle.attributes['role_name'] == ROLE_NAME_IN_XOSC_FOR_EGO_VEHICLE:
                    print(f"Ego vehicle found : role name {vehicle.attributes['role_name']}")
                    vehicle.set_autopilot(True)
                    ego_vehicle = vehicle
                    break

Also I see another issue if I change the following external_control to carla_autopilot as value I get the following error
<AssignControllerAction>
                <Controller name="HeroAgent">
                  <Properties>
                    <Property name="module" value="external_control"/>
                  </Properties>
                </Controller>
              </AssignControllerAction>

File "/home/scenario_runner-0.9.15/srunner/scenariomanager/actorcontrols/actor_control.py", line 95, in __init__
    self.control_instance = getattr(module_control, control_class_name)(actor, args)
AttributeError: module 'carla_autopilot' has no attribute 'CarlaAutopilot'
module 'carla_autopilot' has no attribute 'CarlaAutopilot'
Destroying ego vehicle 215