carla-simulator/scenario_runner #496: Question about build a self-defined scenario

Vela-zz, 2020-03-24:
I want to customize "**scenario4**" in **Route-based scenario**, which correspond to _VehicleTurningRoute_  class under _object_crash_intersection.py_, when I try to figured out how this system works,
I found something that make me confused:

1.
           try:
                self._other_actor_transform = get_opponent_transform(added_dist, waypoint, self._trigger_location)
                first_vehicle = CarlaActorPool.request_new_actor('vehicle.diamondback.century',
                                                                 self._other_actor_transform)
                first_vehicle.set_simulate_physics(enabled=False)
                break
         ......
        # Set the transform to -500 z after we are able to spawn it
        actor_transform = carla.Transform(
            carla.Location(self._other_actor_transform.location.x,
                           self._other_actor_transform.location.y,
                           self._other_actor_transform.location.z - 500),
            self._other_actor_transform.rotation)
        first_vehicle.set_transform(actor_transform)
        first_vehicle.set_simulate_physics(enabled=False)
        self.other_actors.append(first_vehicle)

Why in scenario_runner, it spawn a actor and then set z-axis to - 500 ?
2.
When I try to use this two simple file to build a simplest **scenario4** in **Town1** buy using following file
`<?xml version="1.0" encoding="UTF-8"?>
<routes>
   <route id="0" map="Town01">
      <waypoint pitch="0.0" roll="0.0" x="335.40200805664062" y="180.57827758789062" yaw="90.03459930419922" z="1.0" />
      <waypoint pitch="360.0.0" roll="0.0" x="310.0505142211914" y="195.29856872558594" yaw="180.00465393066406" z="1.0" />
   </route>
</routes>`

 and 

`{
    "available_scenarios": [
        {
            "Town01": [
                {
                    "available_event_configurations": [
                        {
                            "other_actors": {
                                "left": [
                                    {
                                        "pitch": "0.0",
                                        "x": "327.4",
                                        "y": "188.0",
                                        "yaw": "0.0",
                                        "z": "1.0"
                                    }
                                ]
                            },
                            "transform": {
                                "pitch": "0",
                                "x": "330.0",
                                "y": "193.0",
                                "yaw": "150",
                                "z": "1.0"
                            }
                        }
                    ],
                    "scenario_type": "Scenario4"
                }
    ]}]
}`

I deleted all yaw constraints to make it easy to build a scenario, but i found that if z-axis in Question 1 should minus 500, the other actor will born but then disappear and never appear again, even if the ego pass the trigger. 
and if the z-axis do not minus 500, the other actor will stay in correct position but  can not move.

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.8]
 - Python version [e.g. 3.5]

fabianoboril, 2020-03-24:
Hi,
the z-value is set to -500 to enable route-based scenarios. Imagine all traffic participants would be spawned directly at surface level. This would quickly result in a traffic jam. Hence, the actors are under the surface, until they become relevant. As soon as this is the case, they are moved to surface level (usually z=0). If you want to run a single scenario only, you can remove all this parts, or better directly go with OpenSCENARIO.

Vela-zz, 2020-03-25:
> Hi,
> the z-value is set to -500 to enable route-based scenarios. Imagine all traffic participants would be spawned directly at surface level. This would quickly result in a traffic jam. Hence, the [...]

Thank you for you help, 
There are a lot of legal born point checking and adjusting processes before spawn another actor, I delete all of that, then it works fine now.

closed 2020-03-25 by Vela-zz