carla-simulator/scenario_runner #499: Setting ego vehicle speed in manual control mode to a custom value

HarnarayanSingh, 2020-03-28:
How can i set ego vehicle speed in FollowLeadingVehicle scenario for manual_control autopilot mode on?

**Desktop (please complete the following information):**
 - OS: Windows
 - CARLA Version: CARLA 0.9.5
 - Python version :3.7
 - Scenario Runner Version :0.9.5

AMRITKV88, 2020-03-28:
Hi, I tried --autopilot command and it works for me. You can try this command.

Thanks

HarnarayanSingh, 2020-03-28:
> Hi, I tried --autopilot command and it works for me. You can try this command.
> 
> Thanks

I mean I want to change the speed of autopilot in manual_control. By default, it's 20 km/h. How can i change it to a desired value?

fabianoboril, 2020-03-30:
You cannot change the speed of the autopilot. As far as I know, this max. speed is inferred from the underlying map. I'll move this to the CARLA channel, as you might get more help there.