carla-simulator/scenario_runner #502: Actor vehicles losing control in left turn scenario

JeffTheHacker, 2020-04-02:
Question description: 
In the scenario Signalized-Left-Turn, the actor vehicles are being spawned in a stream as I control the ego vehicle trying to make an unprotected left turn. However, the actor vehicles seem to lose control after several seconds. 

Expected behavior: 
The actor vehicles are supposed to move in a straight line across the intersection. However, when I run the scenario, the actor vehicles move straight for about two seconds, and then they lose control. One actor vehicle deviates from its straight path and comes crashing at my ego vehicle. The other actor starts swirling in circles. 

Screenshot: depicts the moment when the actor vehicle hits me 
[image]

Desktop
 - OS: Ubuntu 18.04
 - CARLA Version: 0.9.8
 - Python version: 3.5
 - Version: 0.9.8

fabianoboril, 2020-04-06:
Hi,
I can reproduce the problem. Let's see, if this can be fixed.

fabianoboril, 2020-04-06:
I created a fix for this issue. Please see PR #504

PR #504 (2020-04-07): Fixed WaypointFollower atomic when creating additional LocalPlanners on-the-fly
Fixes #502 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16.04
  * **Python version(s):** 2.7
  * **CARLA version:** 0.9.8

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/504)
closed 2020-04-07 by fpasch