carla-simulator/scenario_runner #507: CONTROLLING EGO VEHICLE via scenario_runner.py or OpenScenario format

iremonur, 2020-04-08:
I want to control the ego vehicle (like give an initial speed or the route to follow for ego vehicle) during the scenario.  In OpenScenario format,  I tried to give an initial speed within <Storyboard> --> <Init> --> <Actions> --> <Private object="hero"> --> <Action> --> <Longitudinal> --> <Speed> --> <Dynamics shape="step"/> --> <Target> --> <Absolute value="10"/>.
But it doesn't work. I think that i can define an initial speed for ego vehicle (hero) in scenario.runner.py but i did not figure it out yet.  Is it possible that give an initial speed or spesific route to ego vehicle in scenario_runner.py or via OpenScenario format (in .xml file) ?

glopezdiest, 2020-04-09:
Hello. **_Speaking only for the scenario_runner.py part_**, you can definitely send a route to the ego_vehicle, it just requires a bit of setting up. Try to run this example and see if it fits your needs:

`python3 scenario_runner.py --route srunner/challenge/routes_training.xml srunner/challenge/all_towns_traffic_scenarios.json 0 --agent srunner/challenge/autoagents/npc_agent.py --debug`

The first part after the route argument is an .xml file that defines the route. The second one is the definition of the scenario, in .json format and the third, the id of the route to be run (if this is blank, they are all sequentially run). Lastly, the agent is the responsible of moving the agent and the debug argument just prints the route (and prints a lot of stuff too).

Regarding the speed, there isn't a parameter to do so right now. If you want to make the ego vehicle start with a specific speed, you'd have to go to the .py file where the scenario in question is defined, and add a pytrees behavior to do so (you can go to srunner/scenariomanager/scenarioatomics/atomic_behaviors.py to see lots of examples)

fabianoboril, 2020-04-30:
That is in fact a bug. Fixed with #519

PR #519 (2020-04-30): Fixed initial speed settings for vehicles in OpenSCENARIO
Fixes #507

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/519)
closed 2020-04-30 by fpasch