carla-simulator/scenario_runner #510: Remaking CarlaDataProvider and CarlaActorPool into a non-singleton.

Ajk4, 2020-04-10:
**Is your feature request related to a problem? Please describe.**

Me and my team are working with reinforcement learning and we use CARLA in our training environment. We would like to use carla-simulator/scenario_runner to specify and run scenarios that our agents learn.

In our pipeline, one trainer node connects to many worker nodes - each running CARLA simulator. It means that within one trainer node we have multiple carla.Client objects and multiple carla.World objects at once.

Current design of having a static singleton makes it impossible/hard to use scenario_runner in our project.

**Describe the solution you'd like**

We could make ScenarioManager object accept instances of CarlaDataProvider and CarlaActorPool in the constructor. Those objects could encapsulate world/actors on a class instance level instead of static level.

Those instances could be passed down to Scenario/AtomicBehaviour objects too when necessary.

**Describe alternatives you've considered**

Currently we have our own code for specifying and running CARLA scenarios.

**Additional context**

I could implement this feature and submit a PR if you are willing to accept it. Please let me know if you are interested.

glopezdiest, 2020-04-24:
Hello @Ajk4 and sorry for the late reply. I was planning on giving you a yes or no answer regarding our interest in this change but at the end, it doesn't make sense, as I haven't even seen your code.

Truth is that we are taking a look at CarlaDataProvider and CarlaActorPool, as they are relatively old and we don't like how they are currently implemented so this comes just at the right time. 

For us, PRs don't have to be fully fleshed additions with no flaws whatsoever so I'd suggest to open a PR and we'll discuss in detail whether or not it is a feature we can incorporate. I can't tell you now if it's gonna be merged but we don't have to decide that right now.

In any case, regardless of whether or not the PR is accepted, thank you very much for wanting to contribute with us, I can tell you right now that it is for sure really appreciated.

Ajk4, 2020-04-24:
Well have that in mind. For know we probably will stick to our custom scenario codes. If necessary we'll open a PR!

Thanks

closed 2020-04-24 by Ajk4

starrye, 2024-05-14:
@Ajk4 Hello, I have a similar problem, do you have any good solutions?