carla-simulator/scenario_runner #514: Data recording during scenario run

semmelknoedel, 2020-04-27:
Is it already possible to record scenario data and store it in a readable file like csv or something similar?
I mean data like actors'postion, yaw, velocity, acceleration, steering angle, timestamp etc?
This data could be very useful to evaluate a scenario outcome.

glopezdiest, 2020-04-27:
Hello @semmelknoedel. Can you specify a little bit more about the objective of this data? Is it meant to be human or program-readable? From your choices I'd assume you want to graph a certain actor attribute over time?

You might be interested in the CARLA recorder feature. It is handled from the 
 [carla.Client](https://carla.readthedocs.io/en/latest/python_api/#carlaclient) class, some of the functions are *start_recorder*, *stop_recorder*, *replay_file* to manage it, and *show_recorder_collisions* and others to extract the results.

If this doesn't suit you, just let me know and we can find another way

semmelknoedel, 2020-04-28:
> Hello @semmelknoedel. Can you specify a little bit more about the objective of this data? Is it meant to be human or program-readable? From your choices I'd assume you want to graph a certain actor attribute over time?

That is exactly what I'm looking for. I want to have different information, position, yaw, etc. and be able to plot them and to use them for calculations.

> You might be interested in the CARLA recorder feature. It is handled from the
> [carla.Client](https://carla.readthedocs.io/en/latest/python_api/#carlaclient) class, some of the functions are [...]

I will check this out and either close this issue or give you more specifications.

closed 2020-05-14 by semmelknoedel

HarnarayanSingh, 2020-05-25:
@semmelknoedel Were you able to log data in csv or any other format? I wish to perform similar with scenario runs recording actor related attributes

semmelknoedel, 2020-05-28:
> @semmelknoedel Were you able to log data in csv or any other format? I wish to perform similar with scenario runs recording actor related attributes

We implemented an extra client that also waits for the ego and then starts recording our relevant data in an csv file each cycle. You have to start it after the scenario but before the ego client.

ParthAD3, 2021-05-05:
> We implemented an extra client that also waits for the ego and then starts recording our relevant data in an csv file each cycle. You have to start it after the scenario but before the ego client.

hey, i was also looking for the same thing, can you share your code which you modified if possible?

semmelknoedel, 2021-05-10:
@ParthAD3 : We don't use it anymore. I recommend to use the metric module. It is very simple to write a class to record the data you need in any format you need.

VidaAPI, 2023-06-19:
Hello @glopezdiest 
Firstly, thanks for the explanation.
can you also guide me through my issue:

I'm trying to record the velocity, acceleration and time of the lead vehicle in follow leading vehicle example scenario. I have modified the behavior class of keepvelocity, and i want to see the results in plots and figures.
Most of the explanations were for the ego vehicle, and i cant find my answer here ( https://carla.readthedocs.io/en/0.9.6/recorder_and_playback/#recording ). I'm relatively new to scenario runner. I couldn't find any related examples for my issue either. Your assistance and guides is appreciated:

I'm using:
CARLA_0.9.13 package version
9.13 => scenario_runner-master
and Windows 10 Enterprise