carla-simulator/scenario_runner #515: Openscenario AttributeError: No ego Vehicles Defined in this scenario

anna-srl, 2020-04-27:
Hi all!

I have been trying to test the overtaking scenario uploaded in #324, using scenario_runner and openscenario:

`<?xml version="1.0" encoding="utf-8"?>
<OpenSCENARIO>
    
<FileHeader revMajor="0" revMinor="9" date="2017-02-24T10:00:00" description="Sample Scenario - Overtaker" author="Andreas Biehn"/>

<ParameterDeclaration/>

<Catalogs>
    <VehicleCatalog>
        <Directory path="Catalogs/VehicleCatalogs"/>
    </VehicleCatalog>
    <DriverCatalog>
        <Directory path="Catalogs/DriverCatalogs"/>
    </DriverCatalog>
    <PedestrianCatalog>
        <Directory path="Catalogs/ObserverCatalogs"/>
    </PedestrianCatalog>
    <PedestrianControllerCatalog>
        <Directory path="Catalogs/PedestrianCatalogs"/>
    </PedestrianControllerCatalog>
    <MiscObjectCatalog>
        <Directory path="Catalogs/MiscObjectCatalogs"/>
    </MiscObjectCatalog>
    <EnvironmentCatalog>
        <Directory path="Catalogs/EnvironmentCatalogs"/>
    </EnvironmentCatalog>
    <ManeuverCatalog>
        <Directory path="Catalogs/ManeuverCatalogs"/>
    </ManeuverCatalog>
    <TrajectoryCatalog>
        <Directory path="Catalogs/TrajectoryCatalog"/>
    </TrajectoryCatalog>
    <RouteCatalog>
        <Directory path="Catalogs/RoutingCatalog"/>
    </RouteCatalog>
</Catalogs>

<RoadNetwork>
    <Logics filepath="Databases/PEGASUS/PEGASUS_A01.xodr"/>
    <SceneGraph filepath="Databases/PEGASUS/PEGASUS_A01.opt.osgb"/>
</RoadNetwork>

<Entities>
    <Object name="Ego">
        <CatalogReference catalogName="VechicleCatalog" entryName="AudiA3_blue_147kW"/>
        <Controller>
            <CatalogReference catalogName="DriverCatalog" entryName="DefaultDriver"/>
        </Controller>
    </Object>
    <Object name="A1">
        <CatalogReference catalogName="VechicleCatalog" entryName="AudiA4_red_147kW"/>
        <Controller>
            <CatalogReference catalogName="DriverCatalog" entryName="DefaultDriver"/>
        </Controller>
    </Object>
</Entities>

<Storyboard>
    <Init>
        <Actions>
            <Private object="Ego">
                <Action>
                    <Longitudinal>
                        <Speed>
                            <Dynamics shape="step"/>
                            <Target>
[... 125 lines cut ...]
> Traceback (most recent call last):
>   File "scenario_runner.py", line 413, in <module>
>     SCENARIORUNNER.run(ARGUMENTS)
>   File "scenario_runner.py", line 224, in run
>     self.run_openscenario(args)
>   File "scenario_runner.py", line 311, in run_openscenario
>     config = OpenScenarioConfiguration(args.openscenario)
>   File "C:\..\scenario_runner-0.9.5.1\srunner\tools\openscenario_parser.py", line 48, in __init__
>     self._parse_openscenario_configuration()
>   File "C:\..\scenario_runner-0.9.5.1\srunner\tools\openscenario_parser.py", line 68, in _parse_openscenario_configuration
>     self._validate_result()
>   File "C:\..\scenario_runner-0.9.5.1\srunner\tools\openscenario_parser.py", line 142, in _validate_result
>     raise AttributeError("No ego vehicles defined in scenario")
> AttributeError: No ego vehicles defined in scenario
> 

Any idea what may be the problem?

Many thanks in advance!

Desktop:
 - OS: Windows 10
 - CARLA Version 0.9.5
 - Python version 3.7
 -Scenario runner version: 0.9.5.1

fabianoboril, 2020-04-30:
Hi,
in this version of SR it was required to have a vehicle flagged as ego vehicle. This you can do by adding a property to the object entity.

```
<Properties>
  <Property name="type" value="ego_vehicle" />
</Properties>
```

Another option is to update to CARLA 0.9.9 and SR 0.9.9, which added support for scenarios with no ego_vehicles.

anna-srl, 2020-05-01:
I changed the code as you suggested and it seems to work, many thanks! 
However, now I keep getting the following error:

>  Error: Unable to spawn vehicle vehicle.lincoln.mkz2017 at Transform(Location(x=0, y=0, z=0.2), Rotation(pitch=0, yaw=0, roll=0))

Does this mean that the waypoint is occupied?  I used the following as a starting position and it doesn't match the one mentioned in the error message:

```
<Private object="Ego">
                <Action>
                    <Longitudinal>
                        <Speed>
                            <Dynamics shape="step"/>
                            <Target>
                                <Absolute value="3.6111111111111107e+01"/>
                            </Target>
                        </Speed>
                    </Longitudinal>
                </Action>
                <Action>
                    <Position>
                        <World x="-9.4" y="-152.8" z="0.5" h="1.57079632679"/>
                    </Position>
                </Action>
            </Private>
```
Any ideas?

fabianoboril, 2020-05-04:
When you get this message, it means there is a collision at the spawn location. This can either be due to another traffic object, or static objects such as houses, lamps etc. I suggest to try another point, and check for example with manual_control to see if the location you have chosen is available.

closed 2020-06-29 by fabianoboril