carla-simulator/scenario_runner #516: I would like to have a vehicle to follow a set of waypoints which should be supported through the tag and the waypoint following atomic behavior. But i can't get the vehicle to start to follow waypoints. I added 10 waypoints. Here is my openscenario.xml file :

iremonur, 2020-04-28:
**Describe our question or idea**
A clear and concise description.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 16.04]
 - CARLA Version [e.g. CARLA 0.9.6]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.6]

iremonur, 2020-04-28:
I would like to a vehicle ("adversary") to follow a set of waypoints which should be supported through the tag in the xml file and the waypoint following atomic behavior. But i can't get my car to start to follow waypoints, the vehicle did not even move.  With this spesification, i expect the vehicle to start following waypints:
 				<Sequence name="RoutingSequence" numberOfExecutions="1">
					<Actors>
						<Entity name="adversary" />
					</Actors>
					<Maneuver name="FollowingRouteManeuver">
						<Event name="FollowsRoute" priority="overwrite">
							<Action name="FollowsRoute">
								<Private>
									<Routing>
										<FollowRoute>
											<Route name="StraightRoute" closed="false">
												<ParameterDeclaration />
												<Waypoint strategy="fastest">
													<Position>
														<World x="-16.0" y="-188.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-15.0" y="-180.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-14.0" y="-170.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-14.0" y="-160.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-13.0" y="-150.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-12.0" y="-140.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-11.0" y="-130.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-10.0" y="-120.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-9.4" y="-100.0" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
												<Waypoint strategy="fastest">
													<Position>
														<World x="-9.4" y="-80" z="0.5" h="1.57079632679" />
													</Position>
												</Waypoint>
											</Route>
										</FollowRoute>
									</Routing>
								</Private>
							</Action>
							<StartConditions>
								<ConditionGroup>
									<Condition name="StartCondition" delay="0" edge="rising">
										<ByValue>
											<SimulationTime value="0" rule="equal_to" />
										</ByValue>
									</Condition>
								</ConditionGroup>
							</StartConditions>
						</Event>
					</Maneuver>
				</Sequence>
NOTE : I cloned latest atomic_behaviour.py
Ubuntu 16.04, CARLA 0.9.7
I would be glad if you help me about this.

fabianoboril, 2020-04-30:
The route itself looks fine, but the start condition might be a blocking factor. Having a starttime of time equal_to 0, is likely to fail, I guess. You can run with "--debug" to get the behavior tree printed, to see where the problem occurs.

iremonur, 2020-04-30:
Thank you for your response. 
I tried to have a starttime of time "greater_than 0" and <SimulationTime value="1" rule="equal_to"/>. 
Both of them didn't work, too. So, i would like to try running with --debug but i use scenario runner with ros. So i run these following commands to execute scenario : 
--> roslaunch carla_ros_scenario_runner carla_ros_scenario_runner.launch
--> rosservice call /scenario_runner/execute_scenario "{ 'scenario': { 'scenario_file': '/home/user_name/carla/scenario_runner-master/srunner/examples/FollowRoute.xml'  } }"
--> python no_rendering_mode.py
So, where should i run the command "--debug" ?

iremonur, 2020-05-01:
I tried to give a start condition which depends on RelativeDistance and it worked. But 4-5 seconds after simulation starts, the vehicle which followed the route("adversary") disappeared. I tried to give "time" specification as you can see below : 
Waypoint strategy="fastest" time = "100"
Position
World x="14.0" y="400.0" z="0.5" h="1.57079632679" /
/Position
But it didn't work.
Is there any way that the "adversary" vehicle does not disappear during following a set of waypoints?

fabianoboril, 2020-05-04:
Hi,
The Resetting/Destroying messages come from CARLA directly, you can ignore these. More likely the scenario is finished, and that is why the vehicle disappears. You can for example run with "--output" to get a scenario summary in stdout.

iremonur, 2020-05-05:
Thank you for your help, i add more waypoints and it worked.

fabianoboril, 2020-05-05:
Excellent :)

closed 2020-05-05 by fabianoboril