carla-simulator/scenario_runner #524: OpenSCENARIO Support - (Relative)RoadPosition and RoutePosition Support

fabianoboril, 2020-05-04:
Add support for RelativeRoadPosition, RoadPosition and RoutePosition for OpenSCENARIO

RoadPosition
https://releases.asam.net/OpenSCENARIO/1.0.0/Model-Documentation/content/RoadPosition.html

RelativeRoadPosition
https://releases.asam.net/OpenSCENARIO/1.0.0/Model-Documentation/content/RelativeRoadPosition.html

RoutePosition
https://releases.asam.net/OpenSCENARIO/1.0.0/Model-Documentation/content/RoutePosition.html

qreezy, 2020-05-22:
Hi @fabianoboril ,
first off, I wanted to thank you for the great work you're doing with this project!
We recently had a look into CARLA/the scenario runner as we were asked whether it would be possible to use CARLA with scenarios exported from our software. As we're relying on the OpenScenario standard (0.9.1 for now, currently updating to 1.0) in our code, we were wondering what your designated timeline for "full" OpenScenario support is?
While the sample XOSCs you provide alongside the runner are of course working, we were unable to successfully execute any of the official examples provided with the OpenScenario 1.0 standard.
Looking forward to your answer and thanks in advance,
Christoph

fabianoboril, 2020-05-25:
Hi @qreezy 
As you can see from the current issues, we made a good start but there is still some work in front of us. Could you let us know, which parts you believe are more critical? Then we might be able to get something working for you earlier than originally planned.
I tested one or two of the official scenarios a while ago, and things were looking good, but I haven't checked if something was updated or if there are more. I will revisit the provided examples.

fabianoboril, 2020-05-26:
I just opened an issue to track the current state of the example scenarios. Please feel free to report your issues in #545

qreezy, 2020-05-26:
Hi @fabianoboril,
I was actually just compiling a list of issues with the official examples and will have a look at issue #545 to post my comments.
In general, the most important features from our point of view would be:
- Private FollowTrajectory actions
- (Relative) RoadPosition
- Full support of ParameterDeclarations
(- Robust XODR-Imports without CARLA crashes <-- but this is probably not a problem of the ScenarioRunner)