carla-simulator/scenario_runner #528: GIVE CUT IN DURATION AND ACCELERATION TO VEHICLE

iremonur, 2020-05-05:
Hi,  i would like to give a specific maneuver duration to vehicle for lane-change maneuver. Is there any way to assign a maneuver duration to it ? 
On the other hand,  I can give an initial speed to ego_vehicle but at the same time i would like to give an acceleration to it. In openscenario_parser.py, it is written that "Acceleration is not implemented yet. " Is there another way to give an acceleration to ego_vehicle ?

Ubuntu 16.04, Carla 0.9.8, Python 2.7

glopezdiest, 2020-05-07:
Hello @iremonur 
> specific maneuver duration to vehicle for lane-change maneuver

What do you mean here? You want to the lane change action to take an specific amount of time?

> Is there another way to give an acceleration to ego_vehicle

We currently have no atomic to set the acceleration to an specific value so you'd have to create one by yourself. It shouldn't be very hard though, here is the atomic that sets the initial velocity (it is in SR 0.9.9 though). With a quick glance, I'd say you just have to change the _set_velocity_ for _add_impulse_.

[image]

glopezdiest, 2020-06-26:
Closing this issue as a TriggerAcceleration atomic has been created, and support to AccelerationCondition

closed 2020-06-26 by glopezdiest