carla-simulator/scenario_runner #534: Automatic control the ego vehicle within the scenario

AkramKalaee, 2020-05-15:
Hello Carla Team,

I've started the scenario runner as follows:
_python scenario_runner.py --scenario  DynamicObjectCrossing_1   --output  --port 7000 --reloadWorld --waitForEgo_

To control the ego vehicle within the scenario, I've opened another terminal and run:
_python automatic_control.py --port 7000 --filter vehicle.tesla.model3_

The problem is that only the ego vehicle is visible and the simulator does not show other actors e.g., pedestrian and vending machine. However, the runner output records the actors' information.

**Screenshots**
[image]

[image]

 - OS: win10
 - CARLA Version: CARLA 0.9.9
 - Python version: 3.7
 - Version: 0.9.9

glopezdiest, 2020-05-20:
Hey @AkramKalaee, I just tried this and it does seem to be an issue. I'll do some testing and give you more information.

glopezdiest, 2020-05-20:
Just checked and indeed, the combination of reloadWorld and waitForEgo had a weird interaction. Did a PR about it.

AkramKalaee, 2020-05-20:
> Hey @AkramKalaee, I just tried this and it does seem to be an issue. I'll do some testing and give you more information.

Thank you very much indeed. Here is the py-trees output in each tick:

[image]

However, the py-trees for manual control is as follows:

[image]

glopezdiest, 2020-05-20:
yeah, the behavior tree stays at *TimeToArrival*, as the world is wrongly initialized, so the scenario is never triggered (however, it is indeed initialized, which is why you can see the actors with the *output* argument).

Please, do try with the changes made at the PR and tell me if it works for you. It's a very small change so you can just copy it directly.

AkramKalaee, 2020-05-20:
Thank you @glopezdiest. I've just tried PR. Now, the other actors are visible but the spawning point of ego is not synced.

[image]

glopezdiest, 2020-05-20:
Okay lemme add a comment, cause I didn't know that either. Apparently, the waitForEgo isn't meant to be used while reloading the world. Basically, it is only meant to work when the ego vehicle is already there, and by reloading the world, everything is removed, so yeah, no more ego vehicle.

Now what is actually your objective here? Why do you want to use the automatic control?

AkramKalaee, 2020-05-20:
I've just checked scenario_runner_0.9.8. It works correctly with automatic control.

> Now what is actually your objective here? Why do you want to use the automatic control?

My aim is to adversarially test the behavior of an AI agent (e.g., automatic control) in some generated situations and find test scenarios leading to failures (e.g., collision).

glopezdiest, 2020-05-20:
> I've just checked scenario_runner_0.9.8. It works correctly with automatic control.

Good to hear! I'll do some more testing then, and we'll see where this goes

glopezdiest, 2020-05-25:
Hey, just wanted to let you know that in the end, we won't be changing the waitForEgo argument. It was a mistake from my part (I thought it worked in another way). I'm also closing this PR as your issue seems to be resolved.

closed 2020-05-25 by glopezdiest