carla-simulator/scenario_runner #544: OpenSCENARIO support in Scenario Runner

HarnarayanSingh, 2020-05-25:
Can we run any OpenSCENARIO file in scenario runner? Or is it just those based on one of the avaialble town maps? 

I have some .xosc scenario files which i want to run.

 - OS: Ubuntu 18.04
 - CARLA Version 0.9.7
 - Python version 3.7
 - Scenario runner 0.9.7

fabianoboril, 2020-05-26:
Hi,
this depends on the scenarios that you have. If these are not based in CARLA towns, you have to use at least the 0.9.8 versions, as otherwise the support of 3rd party OpenDRIVE maps is not given. Finally, it also depends on the use of actions and conditions, some are not yet supported:
https://github.com/carla-simulator/scenario_runner/blob/master/Docs/openscenario_support.md

HarnarayanSingh, 2020-05-26:
Thanks. Would the procedure for simulating be same. In case the actions/ conditions are supported and I'm using version at least 0.9.8, should I just specify my .xosc file path for simulating as specified in readme for available scenarios?

fabianoboril, 2020-05-26:
Yes. This is the way to go. Please also see #545 to get an idea for possible issues.

HarnarayanSingh, 2020-06-07:
@fabianoboril Is there any example of Openscenario file for Scenario runnner that is not based on existing Carla Town Maps?

fabianoboril, 2020-06-29:
No, but you can for example use the once officially provided by ASAM. Alternatively, you can provide the path to the CARLA OpenDRIVE maps instead of the Town-Name.

closed 2020-06-29 by fabianoboril