carla-simulator/scenario_runner #545: Extend and enable support of OSC 1.0 example scenarios

fabianoboril, 2020-05-26:
The OSC 1.0 release comes with several example scenarios, of which some work (partially), others do not work at all.

- [ ] CutIn:
  * Missing Trajectory support
- [ ] EndofTrafficJamNeighboringLaneOccupied
- [ ] FastOvertakeWithReInitialization
- [ ] SlowPrecedingVehicle
  * Sequence of actors with relative positions not handled correctly
- [ ] TrafficJam
  * Scenario broken (Catalogs missing)
- [ ] DoubleLaneChanger
- [ ] EndOfTrafficJam
- [ ] Overtaker
- [ ] SynchronizedArrivalToIntersection

qreezy, 2020-05-26:
EndOfTrafficJam: 
Here the path to XODR is placed in a ParameterDeclaration:
<ParameterDeclaration name="$Road_Logics" parameterType="string" value="Databases/AB_RQ31_Crest.xodr"/>
...
<LogicFile filepath="$Road_Logics"/>

Therefore, the following fails (in openscenario_configuration, _set_carla_town):
"if self.town is not None and ".xodr" in self.town:"
as self.town is "$Road_Logics"

qreezy, 2020-05-26:
EndOfTrafficJamNeighboringLaneOccupied:
Here, first the following fails in openscenario_parser, convert_position_to_transform:
`if dlane == 0:
                    wp = relative_waypoint
                elif dlane == -1:
                    wp = relative_waypoint.get_left_lane()
                elif dlane == 1:
                    wp = relative_waypoint.get_right_lane()
                if wp is None:
                    raise AttributeError("Object '{}' position with dLane={} is not valid".format(obj, dlane))`

because wp will not be defined if dlane is neither of -1;0;1.
Once this is fixed, the proper
`AttributeError: Object 'Ego' position with dLane=2.0 is not valid`
gets thrown, which is due to the fact that changes over more than one lane are currently not supported.

qreezy, 2020-05-26:
DoubleLaneChanger/FastOvertakeWithReInitialization/Overtaker/TrafficJam:
CARLA hard-crashes trying to load SampleDatabase.xodr

qreezy, 2020-05-26:
SynchronizedArrivalToIntersection:
Relative paths to Vehicle/RouteCatalog not supported

fabianoboril, 2020-05-26:
> EndOfTrafficJam:
> Here the path to XODR is placed in a ParameterDeclaration:
> 
> ```
> <ParameterDeclaration name="$Road_Logics" parameterType="string" value="Databases/AB_RQ31_Crest.xodr"/>
> ...
> <LogicFile filepath="$Road_Logics"/>
> ```
> 
> Therefore, the following fails (in openscenario_configuration, _set_carla_town):
> `if self.town is not None and ".xodr" in self.town:`
> as self.town is "$Road_Logics"
> 
> EDIT: Similar for "SlowPrecedingVehicle"

This should be resolved with PR #543

fabianoboril, 2020-05-26:
> DoubleLaneChanger/FastOvertakeWithReInitialization/Overtaker/TrafficJam:
> CARLA hard-crashes trying to load SampleDatabase.xodr

Let me forward this to the CARLA guys who work on the map stuff...

fabianoboril, 2020-06-22:
@qreezy The crash of CARLA when loading certain maps should be resolved with the bug-fix release 0.9.9.4.