carla-simulator/scenario_runner #554: How to run manual_control.py into scenario_manager.py

AMRITKV88, 2020-06-02:
Hi,
To run a openscenario, right now I am running the scenario_runner.py file in one terminal and the manual_control.py (with autopilot mode) in another terminal. So, I called the game_loop() method (with proper arguments) of manual_control.py from scenario_manager.py (where actually simulation starts). Now I can run the scenario, reading from a openscenario file, only once. Because the pygame window keeps running the ego vehicle, and not closing the pygame window though scenario is over. I have to clos ethe pygame window manually. I have tried all such sys.exit(), pygame.display.quit(), and all but there is no positive result.

So, is there any way to merge both the manual_control.py and scenario_runner.py ? Or if you have any other idea to implement this functionality, so that I can run multiple times, the same scenario with slight modification in input parameters in openscenario file, it would be a great help.

Desktop (please complete the following information):
 - OS: [e.g. Ubuntu 16.04]
 - CARLA Version [e.g. CARLA 0.9.8]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.8]

glopezdiest, 2020-06-02:
Hello @AMRITKV88, could you please share the commands you are using to run these scenarios? If I understand correctly, you modified the _scenario_manager.py_ so that it automatically calls the manual control, correct? Could you share that file too?

A couple of doubts. Theoretically, the *manual_control.py* stops when the ego vehicle is destroyed, so is the ego vehicle correctly destroyed at the end of the scenario? Additionally, how are you changing these input parameters at the openscenario file?

Despite all of this, your problem might be fixed by using the *reloadWorld* argument at *scenario_runner.py*, which will for sure close the pygame window as the world is loaded every time a new scenario starts.

AMRITKV88, 2020-06-04:
Hi @glopezdiest, Thanks for your interest ! I am listing the answers to all of your queries below :

**Q1. Ans:** 
Command to run scenario_runner.py : 
       `python3 scenario_runner.py  --output --openscenario /home/ctgpu/software/sdl/scenario_runner/srunner/examples/PedestrianCrossingfront.xosc
`
Command to run manual_control.py :
       `python3 manual_control.py --autopilot
`

**Q2. Ans:** 

[scenario_manager.zip](https://github.com/carla-simulator/scenario_runner/files/4728624/scenario_manager.zip)

I have zipped the scenario_manager.py file. Actually I am extracting the trajectory from the simulation. So, if I am running the scenario_runner file in one terminal and manual_control in another terminal, then its working fine and trajectory also get extracted properly (It is when I am not calling the manual_control file from scenario_manager. I am running both the file in separate terminal). 

Now, the thing is that if I go to call the main of manual_control.py from scenario_manager.py file (you can see in the code, line no:- 226 in scenario_manager.py), as the openscenario file already got loaded, it is only simulating not giving any trajectory. And apart from that its not getting into the while loop, you can see in the scenario_manager.py file's while loop at line no.-  235.

**Q3. Ans:**

As you can see that ego and all other agent vehicles are destroyed at openscenario end, not the manual control end.

**Q4. Ans:**

I have written a method which will load the openscenario file and change the required fields and replace it, before running the simulation.

And to last your suggestion about `--reloadworld`, its possible when I have multiple openscenario files.

My main aim is to run both the manual_control.py and scenario_runner.py into same terminal. So, basically the process flow will be: after modification + loading the openscenario file, when it will get into the run_openscenario method, it will call the manual_control with autopilot mode. After the simulation, it should close the pygame window and I will be able to get the simulation trajectory....! And this process repeats again after modifying the required fields in openscenario file....!

Please let me know if you need more clearification or any further information.

Thanks Again @glopezdiest ....!

glopezdiest, 2020-06-04:
Thanks for all the info @AMRITKV88. Substituting the scenario_manager file to SR 0.9.8, I can see that, as you said, if the manual control is run on another terminal, it is indeed working.

However, if I activate the manual_control through the scenario_manager, that call is blocking the execution of the scenario, and the *watchdog* triggers a timeout (+ I had to remove the arguments as they were causing problems). Why isn't this happening to you? If you can, I'd appreciate if you attached the whole scenario_runner folder so it's easier for me to test.

fabianoboril, 2020-06-04:
There might be a simpler option ;)
We are now supporting OSC controllers. The implementation at the end is located here:
https://github.com/carla-simulator/scenario_runner/tree/master/srunner/scenariomanager/actorcontrols
You can create a "manual_control" and assign it to your ego vehicle. This would allow you to run the only scenario_runner, and the "manual_control" thing will appear everytime you start the scenario, and disappear afterwards.

AMRITKV88, 2020-06-08:
Hi @glopezdiest  !
Thanks for your prompt reply and I am extremely sorry for uneven delay.
Here I have attached the scenario runner file for your reference. 

As  I am new to this CARLA simulator, I am trying all the way possible, but I am stuck at this.

Please help me out.

Thanks again @glopezdiest ....!

[scenario_runner-0.9.8.zip](https://github.com/carla-simulator/scenario_runner/files/4746471/scenario_runner-0.9.8.zip)

AMRITKV88, 2020-06-08:
Hi @fabianoboril  !

I could not understand clearly what you said.

I am new to this carla simulator. So, I would request you to elaborate a bit what you said or if you please share some code snippet, it would be a great help.

Thanks in advance @fabianoboril ....!

glopezdiest, 2020-06-09:
@AMRITKV88.
> And apart from that its not getting into the while loop, you can see in the scenario_manager.py file's while loop at line no.- 235.

You might just be confusing yourself because you've already identified your error. When you call *manual_control.main()*, you are blocking the execution of the scenario manager, so the scenario stops ticking.

You have to run the manual_control in parallel of scenario_manager (which is what you are inherently doing when running it in a separate terminal). The manual_control automatically stops when the ego vehicle is destroyed so this can be easily done by adding a thread that just runs *manual_control.main()*

> **Q3. Ans:**
As you can see that ego and all other agent vehicles are destroyed at openscenario end, not the manual control end.

I actually can't see this, because of the aforementioned issue, but if I manually interrupt the manual control, everything is correctly removed

AMRITKV88, 2020-06-09:
@glopezdiest 

In the terminal where scenario_runner.py has run, in that terminal I have found that the ego is terminated here. Here I have attached the screenshot. Please have a look. 

Actually if you run the openscenario file I have attached herewith, then you don't have to manually terminate the manual_control.py running terminal. This is because, the scenario termination condition is changed a little bit.

So, in that case you might see that scenario is terminated and ego is destroyed in scenario
-runner side.

[OpenSCENARIO.zip](https://github.com/carla-simulator/scenario_runner/files/4750682/OpenSCENARIO.zip)

[image]

glopezdiest, 2020-06-09:
I cant run that file, as it isn't an *.xosc* file (its giving me parsing errors) Can you actually run that?

In any case, the manual control issue should still apply, regardless of the openscenario file. So is your scenario being correctly triggered? It does seem that way seeing the output message .

AMRITKV88, 2020-06-09:
@glopezdiest 
I am extremely sorry for this. Actually I zipped up a wrong file. Here I have attached the correct file.
Again I am really sorry for the wrong file.

Thanks again @glopezdiest ....!
[PedestrianCrossingfront.zip](https://github.com/carla-simulator/scenario_runner/files/4751895/PedestrianCrossingfront.zip)

AMRITKV88, 2020-06-09:
@glopezdiest 

As I mentioned earlier, if I am running in two different terminals then its totally fine. And the output displayed on the open
-scenario running terminal is what I posted above. 

Now if I am trying to call the manual_control.py from scenario_manager file, the scenario runs, as it gets loaded earlier before invoking manual_control from scenario_manager file. But after the termination condition (in my case I have set a particular driven distance for ego, in .xosc file) it is supposed to stop, but the pygame window keeps it running until I manually close the pygame window.

So, when I debug it, in scenario_manager file, the while loop right below the call of manual_control, its not executing. I get to this conclusion because I put a print statement at the very first line in the while loop. And its not printed out.

Please help me !

Thanks @glopezdiest ....!

glopezdiest, 2020-06-09:
> Actually I zipped up a wrong file.

Don't worry, that happens to everyone. 

> So, when I debug it, in scenario_manager file, the while loop right below the call of manual_control, its not executing

I think we are on the same page here. That is indeed what I've been seeing and my previous answer still stands.

If you run the **manual control from another terminal**, it is run **in parallel** to scenario_manager. 
But, if you run **manual control from scenario_manager**, then it is run **in series**.

As you already noticed, this blocks the main loop, which blocks the execution of the scenario. As this scenario isn't being executed, it can never end, which is why the manual control never ends either.

If you want to use the manual control directly from the scenario_manager, you have to do so in parallel, using the *thread* library or anything similar.

AMRITKV88, 2020-06-10:
@glopezdiest

Thanks a lot for your kind help ! :-)

AMRITKV88, 2020-06-10:
@fabianoboril 
 
Can you please help me out in your procedure, what u explained earlier ?

Please help !

Thanks again @fabianoboril ....!

AMRITKV88, 2020-06-23:
Hi @glopezdiest !

I want to know few details about my **question no 4.**

So, the thing is that I am trying to reload the same file with minor changes in some parameters. So, is it possible to repeatedly load the same openscenario file ? Because I am facing issue in that.

The issue is that - After first run of the openscenario file, when its entering into the second run, its getting terminated. I have attached a screenshot of that error. Please have a look.

[image]

Please Help !

glopezdiest, 2020-06-23:
Everytime you run *scenario_runner.py*, an scenario is being triggered, so you just have to run that several times. There are methods to repeat the scenario, but you are changing the OSC file between repetitions, correct?

AMRITKV88, 2020-06-23:
@glopezdiest 

You are absolutely correct !

I am changing few coordinate values of ego, agents and pedestrians in the openscenario (.xosc) file and replacing the earlier file with the new file and with same name.

For example, I am explaining the entire procedure for your better understandings :- 

    1. I have a for loop which runs for say 100 times, 
    2. For first loop, it runs scenario_runner with fresh openscenario (.xosc) file,
    3. Once this openscenario file is executed, it destroys the ego vehicle and scenario terminates,
    4. For second loop at first it will open the openscenario (.xosc) file and make changes to few coordinate values,
    5. Replaces the existing openscenario (.xosc) file with the new updated openscenario (.xosc) file,
    6. Then it starts executing the new openscenairo(.xosc) file.
    7. Again 3rd loop starts similarly.

This is the entire procedure I want to implement.

So, can you please help me out of this !

Thanks in advance !

glopezdiest, 2020-06-26:
Then you just have to do a *while* that has two steps:
1) Run scenario_runner directly from the script
2) Change the OpenSCENARIO file.

If you overwrite the OSC file, then the command to run scenario_runner will be the same, because the file is the same, so no problem there

AMRITKV88, 2020-06-28:
Thanks a lot for your continuous support @glopezdiest !

glopezdiest, 2020-07-01:
@AMRITKV88 Do you have any more doubts or can we consider this issue closed?

AMRITKV88, 2020-07-01:
Hi @glopezdiest !

I have no doubts for the time being. So, we can consider it issue to be closed now.

If anything new comes up, I would definitely reach out to you.

Thanks a lot for your kind help @glopezdiest  !

glopezdiest, 2020-07-01:
My pleasure :+1:

closed 2020-07-01 by glopezdiest