carla-simulator/scenario_runner #555: OPENScenario: ‘priority’ attribute support in Event

aqvia, 2020-06-02:
As far as reading ScenarioRunner's scripts,  in OpenSCENARIO, there doesn't seem to be any support for the ‘priority’ attribute of event. Is there any plan to do so in the future?

**Problem**

In OpenSCENARIO, I tried to create a scenario in which a vehicle decelerates and stops while changing lanes. To do so, it is necessary to run LaneChangeAction and SpeedAction in parallel.

1) As shown below, if listing those actions in parallel in a same event, the car slows down and comes to a stop before changing lanes. This may be because throttle is set to 0.0 from the start of lane change. Therefore, I think I need to delay the start of SpeedAction.

```xml
<Event name="A" priority="overwrite">
  <Action name="A1">
    <PrivateAction>
      <LateralAction>
        <LaneChangeAction>
          <LaneChangeActionDynamics dynamicsShape="linear" value="10.0" dynamicsDimension="distance"/>
          <LaneChangeTarget>
            <RelativeTargetLane entityRef="P" value="-1"/>
          </LaneChangeTarget>
        </LaneChangeAction>
      </LateralAction>
    </PrivateAction>
  </Action>
  <Action name="A2">
    <PrivateAction>
      <LongitudinalAction>
        <SpeedAction>
          <SpeedActionDynamics dynamicsShape="step" value="115.0" dynamicsDimension="distance"/>
          <SpeedActionTarget>
            <AbsoluteTargetSpeed value="0.0"/>
          </SpeedActionTarget>
        </SpeedAction>
      </LongitudinalAction>
    </PrivateAction>
  </Action>
  <StartTrigger>
    <!-- ... -->
  </StartTrigger>
</Event>
```

2) As shown below, if parallelizing Actions as separate events, I have to specify priority="parallel" for the deceleration event.

```xml
<Event name="A" priority="overwrite">
  <Action name="A1">
    <PrivateAction>
      <LateralAction>
        <LaneChangeAction>
          <LaneChangeActionDynamics dynamicsShape="linear" value="10.0" dynamicsDimension="distance"/>
          <LaneChangeTarget>
            <RelativeTargetLane entityRef="P" value="-1"/>
          </LaneChangeTarget>
        </LaneChangeAction>
      </LateralAction>
    </PrivateAction>
  </Action>
  <StartTrigger>
    <!-- ... -->
  </StartTrigger>
</Event>
<Event name="B" priority="parallel">
  <Action name="B1">
    <PrivateAction>
      <LongitudinalAction>
        <SpeedAction>
          <SpeedActionDynamics dynamicsShape="step" value="100.0" dynamicsDimension="distance"/>
          <SpeedActionTarget>
            <AbsoluteTargetSpeed value="0.0"/>
          </SpeedActionTarget>
        </SpeedAction>
      </LongitudinalAction>
    </PrivateAction>
  </Action>
  <StartTrigger>
    <!-- During the lane change action -->
  </StartTrigger>
</Event>
```

fabianoboril, 2020-06-03:
Hi,
could you please test the latest version of SR. We recently reworked the way the entities are controlled, and my assumption is that your lane change should work now as you envisioned it. Having said that, still the priority tag is not evaluated.

aqvia, 2020-06-04:
@fabianoboril Thank you for your reply.
I'm sorry that I left out the original description, but I was running both Carla and ScenarioRunner with the tag 0.9.9 version. Especially, in case (2), if the timing after lane change is not good, the car's trajectory will shift during deceleration action. When I upgraded both to master and ran it, this problem was solved.
However, the priority attribute of Event seems not to be supported, and events do the (default) "Overwrite" movement. Do you have a schedule for this feature?

fabianoboril, 2020-06-04:
@aqvia As said, I think the lane change as you described it may work in master.

Regarding the priority tag: Currently we are working in closing the major gaps like trajectory support or conditions. If you like you may want to take a look yourself, of course with our support.