carla-simulator/scenario_runner #556: unable to parse the OpenDrive XML string

xli4217, 2020-06-02:
**Describe the bug**
Getting error "unable to parse the OpenDrive XML string" when running scenarios

**To Reproduce**
Steps to reproduce the behavior:
1. Start Carla `./CarlaUE4./sh`
2. Run scenario `python scenario_runner.py --scenario OtherLeadingVehicle_1 --reloadWorld
`

**Expected behavior**
Successful running of the specified scenario

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.9.2]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.9]

fabianoboril, 2020-06-03:
Hi,
could you please test the latest master of ScenarioRunner? I checked it with CARLA 0.9.9 and 0.9.9.2 and I cannot reproduce your problem.

xli4217, 2020-06-03:
Tried with master, problem still occurs. I'm using anaconda python 3.7, could this be an issue?

fabianoboril, 2020-06-03:
Is CARLA or ScenarioRunner complaining about the OpenDRIVE issue? Can you post a screenshot?

raymanfx, 2020-06-03:
Same issue here.
I'm running Fedora 32 with python3.7 installed manually.
My CARLA version is 0.9.9.2 and my ScenarioRunner version is 0.9.9.

The CARLA server is complaining about the OpenDRIVE issue.
It appears like the world reload is successful, because I can see a new scene, but there are no vehicles or any kind of NPCs around, nothing happens - you can only move around the camera with WASD keys in the CARLA server UE4 window.

raymanfx, 2020-06-03:
This appears to be the line: https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/opendrive/OpenDriveParser.cpp#L32. Is there a way to get some debug output from the XML parser?

fabianoboril, 2020-06-03:
I see the problem if I run with Python 3.5, but not with Python 2.7. I guess this is a CARLA PythonAPI issue. Could you please open an issue there?

fabianoboril, 2020-06-03:
For the moment I would recommend to use v0.9.9, if this solves the problem (at least here I cannot see the OpenDRIVE error message anymore).

xli4217, 2020-06-03:
Tried with python 2.7 + scenario_runner master branch + carla latest debian (0.9.9.2-1), error still occurs.

raymanfx, 2020-06-03:
Python 2.7 is EOL, so using it is not an option (for me and many others I guess).
I can confirm that CARLA 0.9.9 is fine and scenarios work.

xli4217, 2020-06-03:
Is this 0.9.9 from debian or cloned from repo?

raymanfx, 2020-06-03:
I'm talking about CARLA 0.9.9 from the project GitHub releases. Not built from source.

xli4217, 2020-06-03:
I see. The problem is still there for me. I've opened an issue on carla-simulator.

xli4217, 2020-06-05:
Just to update, I tried with CARLA 0.9.8 and the corresponding scenario_runner, the error disappears and agents (except for ego) spawns, but none of the agents are moving. @raymanfx you had similar problem but then confirmed everything is running, is there something that you modified?

raymanfx, 2020-06-10:
Hi, I didn't have to change anything. I assume you are aware of this, but you have to manually control the ego vehicle and drive towards the spawned agent(s) to trigger the scenario.

fabianoboril, 2020-06-22:
@raymanfx @xli4217 
Please test the new CARLA 0.9.9.4 release. It resolve the problem with the OpenDRIVE warning on my side.

closed 2020-06-29 by fabianoboril