carla-simulator/scenario_runner #562: OpenScenario SynchronizeAction support

gjossep, 2020-06-09:
We are working on using OpenScenario and ScenarioRunner. However we are very interested in using the SynchronizeAction, this is currently not supported.

I was wondering if any work has been done on this, and if possible how we can help. We know its not a trivial task, and can be quite complex. 

We are also interested in creating the support for it our self's and contributing to ScenarioRunner. If this is the case could you point us in the direction to get started?

glopezdiest, 2020-06-10:
Hello @gjossep. As far as I know we currently aren't working on implementing *SynchronizeAction* but I can be wrong so let me tag @fabianoboril in case he knows something I don't. 

> We are also interested in creating the support for it our self's and contributing to ScenarioRunner. 

That's very thoughtful of you! We definitely appreciate any help you can provide.

> If this is the case could you point us in the direction to get started?

As you may know, ScenarioRunner creates the scenarios as behavior trees, and so does OpenSCENARIO. Therefore, to implement *SynchronizeAction*, two steps are needed:

1. Parse the information of the *.xosc* into an atomic ScenarioRunner can understand.
2. Create / Modify / Expand atomics to reproduce *SynchronizeAction*.

For the first part, you can take a look at _srunner/tools/openscenario_parser.py_, where all the OpenSCENARIO semantics are parsed.

For the second one, check out _srunner/scenariomanager/scenarioatomics_ for a very long list of currently implemented atomics. *SynchronizeAction* is considered a behavior, so it might be a good idea to add it to *atomic_behaviors.py*

ThrashAbaddon, 2020-09-23:
> As you may know, ScenarioRunner creates the scenarios as behavior trees, and so does OpenSCENARIO. Therefore, to implement _SynchronizeAction_, two steps are needed:

OpenSCENARIO doesn't use behavior trees. Where did you get that information?

glopezdiest, 2020-12-03:
Hey, as of PR #689, SynchronizeActions are now implemented

closed 2020-12-03 by glopezdiest