carla-simulator/scenario_runner #566: Timeout when running scenario_runner.py

sissiok, 2020-06-17:
I tried to run a scenario with command python3 scenario_runner.py --scenario FollowLeadingVehicle_1 --output  , but failed and got a timeout. I hope someone can help me with this issue.

Below is the output I got:

Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle

Scenario: FollowVehicle --- Result: TIMEOUT
Start time: 2020-06-17 12:02:27
End time: 2020-06-17 12:03:28
Duration: System Time 60.72s --- Game Time 60.59s
Ego vehicle:  Actor(id=789, type=vehicle.lincoln.mkz2017)
Other actors: Actor(id=790, type=vehicle.nissan.patrol); 

                Actor             |            Criterion           |   Result    | Actual Value | Expected Value 
-----------------------------------------------------------------------------------------------------------------
         lincoln.mkz2017 (id=789) |         CheckCollisions (Req.) |     SUCCESS |         0.00 |         0.00 
                                  |                       Duration |     FAILURE |        60.59 |        60.00 

Not all scenario tests were successful
Destroying ego vehicle 789
No more scenarios .... Exiting

 - OS: Ubuntu 18.04
 - CARLA Version: CARLA 0.9.9
 - Python version: 3.6
 - Version: 0.9.9

glopezdiest, 2020-06-18:
Hello @sissiok. Have you used the manual control to trigger the scenario? By default you have to run the scenario in one terminal and the manual_control (or any agent) in another. 

Also, **FollowLeadingVehicle had a bug** that prevented it from correctly finishing (so it was only able to finish by timeout), which we fixed 3 days ago, so that might also be causing the issue.

glopezdiest, 2020-06-26:
@sissiok Was this useful? Can we consider the issue closed?

sissiok, 2020-06-27:
Thanks a lot for the help. Now I know how it works. I did not realize that I should run scenario and manual_control at the same time in different terminals.

closed 2020-06-27 by sissiok

shh1v, 2023-07-23:
Why is it necessary to run `manual_control.py`? What functionality in that file is implemented to trigger the scenario?

glopezdiest, 2023-07-24:
Running `scenario_runner.py` only prepares the scenarios. It is when the ego starts moving that these start running. The ego's  movement can be achieved with any tool you want, manual control, TrafficManager...

Ideally, this movement should be done by the user's AV stack but the examples can't make use of that. A good substitute is to use the manual control, which comes with a camera and a keyboard control, allowing the users themselves to "be the AV stack" moving the ego vehicle.

shh1v, 2023-07-24:
Thank you for the response. This was very helpful @glopezdiest.