carla-simulator/scenario_runner #567: Scenario not yet ready with manual_control.py

sissiok, 2020-06-17:
I tried to run manual_control.py, but only a black window popped out and in the terminal I got message "scenario not yet ready". What went wrong? Is there anyone can help me with it?

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version: CARLA 0.9.9
 - Python version: 3.6
 - Version: 0.9.9

glopezdiest, 2020-06-18:
Nothing went wrong. The manual control at ScenarioRunner doesn't create any vehicles. It instead attaches to an already spawned vehicle which has the "hero" rolename (the ego vehicle). If that isn't the case, the *Scenario not yet ready* will constantly pop up until the vehicle is created

fabianoboril, 2020-06-22:
Copy of #566. Let's close this one, and continue in #566.

closed 2020-06-22 by fabianoboril

sissiok, 2020-06-27:
Thank you. It is solved.