carla-simulator/scenario_runner #572: OpenScenario AttributeError: 'NoneType' object has no attribute 'get_forward_vector'

sehrizadasahinovic, 2020-06-23:
I tried to run:
 python scenario_runner.py --openscenario srunner/examples/PedestrianCrossingFront.xosc --reloadWorld --output 

and scenario started but after  few seconds it stoped with an error:

AttributeError: 'NoneType' object has no attribute 'get_forward_vector'
'NoneType' object has no attribute 'get_forward_vector'
Destroying ego vehicle 2953
No more scenarios .... Exiting

System settings:
    OS: Ubuntu 18.04
    CARLA Version: CARLA 0.9.9
    Python version: 2.7.17
    Version: 0.9.9

glopezdiest, 2020-06-25:
Hey @sehrizadasahinovic wasn't there any kind of traceback information that you can share?

sehrizadasahinovic, 2020-06-25:
@glopezdiest here is the traceback information:

Traceback (most recent call last):
  File "scenario_runner.py", line 382, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/home/sehrizadasahinovic/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 216, in run_scenario
    self._tick_scenario(timestamp)
  File "/home/sehrizadasahinovic/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 260, in _tick_scenario
    self.scenario_tree.tick_once()
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 247, in tick
    self.initialise()
  File "/home/sehrizadasahinovic/scenario_runner/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py", line 484, in initialise
    self._control.direction = CarlaDataProvider.get_transform(self._actor).get_forward_vector()
AttributeError: 'NoneType' object has no attribute 'get_forward_vector'
'NoneType' object has no attribute 'get_forward_vector'
Destroying ego vehicle 433
No more scenarios .... Exiting

glopezdiest, 2020-06-25:
Ah yes, that's actually something I just found yesterday and was meaning to investigate it. From the first inspection, seems like a bug related to walkers not being part of the CarlaDataProvider, so you can't use it to get the transform. A patch would be to change

> self._control.direction = CarlaDataProvider.get_transform(self._actor).get_forward_vector()

for:

> self._control.direction = self._actor.get_transform().get_forward_vector()

Again, I'm not sure this is the exact reason, and I'll investigate it today. I'll let you know what I find.

sehrizadasahinovic, 2020-06-25:
@glopezdiest ok, thank you very much for your support.

sehrizadasahinovic, 2020-06-25:
Now, there is another error:

Traceback (most recent call last):
  File "scenario_runner.py", line 382, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/home/sehrizadasahinovic/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 216, in run_scenario
    self._tick_scenario(timestamp)
  File "/home/sehrizadasahinovic/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 260, in _tick_scenario
    self.scenario_tree.tick_once()
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 309, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/composites.py", line 494, in tick
    for node in child.tick():
  File "/home/sehrizadasahinovic/scenario_runner/srunner/tools/py_trees_port.py", line 61, in tick
    for node in self.decorated.tick():
  File "/home/sehrizadasahinovic/.local/lib/python2.7/site-packages/py_trees/behaviour.py", line 249, in tick
    new_status = self.update()
  File "/home/sehrizadasahinovic/scenario_runner/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py", line 505, in update
    self._distance += calculate_distance(self._location, new_location)
  File "/home/sehrizadasahinovic/scenario_runner/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py", line 54, in calculate_distance
    return location.distance(other_location)
AttributeError: 'NoneType' object has no attribute 'distance'
'NoneType' object has no attribute 'distance'
Destroying ego vehicle 433
No more scenarios .... Exiting

glopezdiest, 2020-06-25:
@sehrizadasahinovic First, this isn't related to the problem I mentioned before, that was due to my own stupidity.

Now, I just tried with CARLA 0.9.9 and it is indeed failing but it works with current CARLA master. Why? CARLA 0.9.9 has a problem related to the loading of the world, which causes some actor instances to be wrong. This was solved at CARLA 0.9.9.3 + a patch to UnrealEngine (patch instructions [here](https://carla.readthedocs.io/en/latest/build_linux/), a t the dependencies section). 

Therefore, you have **two options**:
- Do not reload the world (don't use --reloadWorld)
- Upgrade to CARLA 0.9.9.3 and patch Unreal

sehrizadasahinovic, 2020-06-25:
Hi @glopezdiest , it works now. Thx a lot.

closed 2020-06-25 by glopezdiest