carla-simulator/scenario_runner #577: OpenSCENARIO + Agent support

lucasnmatias, 2020-06-25:
Hi,

I have one question about the support for OpenSCENARIO. I would like to know why we cannot run a scenario from OpenSCENARIO with an agent implemented?

`Agents are currently only compatible with route scenarios`

It would be interesting to define a scenario with OpenSCENARIO and test my agent on it. Also, right now I don't see how to define the ego car on OpenSCENARIO and setup sensors on it, so the OpenSCENARIO does not help me to collect sensor data from the specified scenarios.

fabianoboril, 2020-06-29:
Hi,
you can make use of an agent via OSCControllers (please see the corresponding example here: https://github.com/carla-simulator/scenario_runner/blob/master/srunner/examples/OscControllerExample.xosc)
The controllers are defined here: https://github.com/carla-simulator/scenario_runner/tree/master/srunner/scenariomanager/actorcontrols
You can implement your own (derived from BasicControl) to be your agent. The ego vehicle is simply flag with an ego_vehicle property (see also the OSC examples).

lucasnmatias, 2020-06-29:
That is great! I just missed one thing, different from `BasicAgent`, `BasicControl` does not implement the function `setup_sensors`. I work at IVEX and since we work with assess of autonomous vehicle safety we have been implementing scenarios and using the sensor's output to evaluate vehicle behavior. Is it possible to use `BasicControl` + OpenSCENARIO and setup sensors on the car?

We want to assess the vehicle behavior in two ways with the simulator:

1 - Assess the agent safety with OpenSCENARIO scenes (control the ego with an implemented agent and collect data from sensors)
2 - Implement a specific behavior on the ego vehicle with OpenSCENARIO (control the ego with OpenSCENARIO and also collect data from sensors)

Right now our solution has been a modification on the original code, we remove the if condition that blocks the use of agent + OpenSCENARIO and passes our implemented agent as a parameter normally (so we can setup sensors to the ego car and control it without OpenSCENARIO). Also, we add a new flag `dummyAgent` to allow setup sensors on the ego vehicle and control it through OpenSCENARIO. But I'm trying to understand better the original code to see if we can have those functionalities without these adaptations.

fabianoboril, 2020-07-08:
Setting up sensors can be done through the control as well. You just have to pass the sensors as properties in the XOSC file, or you hard-code them in your control implementation. I did that for the one I am currently using for my own purpose.

SiNa-Dehbari, 2020-11-23:
Hello Everyone,

I have the same challenge.
I am new in Carla and ScenarioRunner. My problem is:  I want to define a scenario in Openscenario Format and run it automatically. (I don't want to use `manual_control.py`
could you please give me the commands?

Thank you in advance