carla-simulator/scenario_runner #584: WARNING:OpenScenarioConfiguration:Wrong OpenDRIVE map in use. Forcing reload of CARLA world

LikhitaNukala, 2020-06-30:
I have executed the following command for OpenSCENARIO.
**python scenario_runner.py --openscenario L:\carla_latest\scenario_runner-0.9.9\srunner\examples\CyclistCrossing.xosc**

I'm getting the following.

WARNING:OpenScenarioConfiguration:Wrong OpenDRIVE map in use. Forcing reload of CARLA world
Preparing scenario: CARLA:CyclistCrossing
ScenarioManager: Running scenario OpenScenario
Traceback (most recent call last):
  File "scenario_runner.py", line 382, in _load_and_run_scenario
    self.manager.run_scenario()
  File "L:\carla_latest\scenario_runner-0.9.9\srunner\scenariomanager\scenario_manager.py", line 216, in run_scenario
    self._tick_scenario(timestamp)
  File "L:\carla_latest\scenario_runner-0.9.9\srunner\scenariomanager\scenario_manager.py", line 260, in _tick_scenario
    self.scenario_tree.tick_once()
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\behaviour.py", line 249, in tick
    new_status = self.update()
  File "L:\carla_latest\scenario_runner-0.9.9\srunner\scenariomanager\scenarioatomics\atomic_criteria.py", line 648, in update
    current_wp = self._map.get_waypoint(current_loc, lane_type=carla.LaneType.Any)
OverflowError: can't convert negative value to unsigned int
can't convert negative value to unsigned int
Destroying ego vehicle 258
No more scenarios .... Exiting

**Desktop **
 - OS: Windows 10
 - CARLA Version CARLA 0.9.9
 - Python version 3.7.6
 - Version  0.9.9

glopezdiest, 2020-06-30:
Hey. That WARNING is normal. Each scenario needs a specific town so if you aren't in the correct one, the world is reloaded. 

> OverflowError: can't convert negative value to unsigned int

This however, isn't. Can you explain the step you did to reproduce it? I just tried it and couldn't see anything like that.

PD: I'm changing the title to match the actual issue

LikhitaNukala, 2020-07-03:
[image]

I have followed the same steps of importing/ simulating OpenSCENARIO files using the command. 
python scenario_runner.py --openscenario <path_of_the_file>
I'm getting the error.

glopezdiest, 2020-07-03:
Okay, I'll take a look, see if I can reproduce the error.

glopezdiest, 2020-07-06:
@LikhitaNukala I'm sorry to say this but I can't seem to reproduce this error. This makes me believe it might be a Windows related issue, but I can't ensure that's the case, as I currently don't have the possibility to test it in Windows. I'll keep digging, see if I find something else.

DevotionGit, 2020-07-14:
I got this error on windows10,not for ubuntu.

glopezdiest, 2020-07-14:
@DevotionGit Could you please print the exact *current_loc* and paste it here?

charlie-wartnaby, 2020-07-22:
Hello, I've just run into this under Windows too. It seems to be because:
```
carla\PythonAPI\carla\dependencies\include\carla\road\Lane.h(51):      Any           = 0xFFFFFFFE
```
... which should be fine, but in Windows Python (Python 3.8.3 (tags/v3.8.3:6f8c832, May 13 2020, 22:37:02) [MSC v.1924 64 bit (AMD64)] on win32), that gets interpreted like a signed int as -2. Or at least does if I use an int() conversion in a debug print statement in the Python.

So then the Python is passing -2 to a function expecting a uint32_t when it uses LaneType.Any here in `npc_vehicle_control.py`:
```
            waypoint = CarlaDataProvider.get_map().get_waypoint(
                transform.location, project_to_road=True, lane_type=carla.LaneType.Any)
```
Changing it to `carla.LaneType.Driving` as  a workaround avoids the exception (in my case at least). But obviously that's not a proper solution.

glopezdiest, 2020-07-23:
Hey @charlie-wartnaby. Thanks for the information. We'll take a look, see what we can do to fix it.

charlie-wartnaby, 2020-07-28:
Thank you @glopezdiest , and note @LikhitaNukala and @DevotionGit . For now a simple workaround is to change the definition of `LaneType::Any` here in `LibCarla/source/carla/road/Lane.h` (and then rebuild CARLA and the PythonAPI):

```
      Any           = 0x7FFFFFFE // CW: IDIADA workaround to avoid overflow exception when 0xFFFFFFFE interpreted as -2 in Windows Python and passed to uint32_t function in API
```

For good measure I did the same thing for 'LightState::All` in `LibCarla/source/rpc/VehicleLightState.h`, I didn't spot any others like this:

```
      All          = 0x7FFFFFFF // CW: IDIADA workaround to avoid overflow exception when 0xFFFFF... interpreted as -ve in Windows Python
```

I have never written my own Python library in C/C++ and am new to CARLA, so to be honest I have no idea how the type mapping is supposed to work there from C++ to Python. But this seems to do the trick.