carla-simulator/scenario_runner #596: Scenario_runner: No mudule named 'xmlschema'

Lu7913, 2020-07-15:
Hello,
I'm new to the Carla and scenario runner and I have the following problem when trying to start an example scenario. 

Running: py -3 scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorldThe 

resulted in error: 
File "scenario_runner.py", line 34, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "...\scenario_runner\srunner\scenarioconfigs\openscenario_configuration.py", line 18, in <module>
    import xmlschema
ModuleNotFoundError: No module named 'xmlschema'

I came to a dead end. 

Also in Getting started tutorial [https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_scenariorunner.md](url) path to .egg file in Windows seems to be not working for me. I found the solution in #580, where is different path specified. 

 - OS: Windows 10
 - CARLA Version 9.9
 - Python version 3.8

Lu7913, 2020-07-16:
I had to install additional python modules.

Lu7913, 2020-07-22:
Could this be related to tensorflow problems which I had, (had to downgrade to 1.15.3)?

glopezdiest, 2020-07-22:
Hey @Lu7913. That's definitely related to the PYTHONPATH. Could you please post it here? Theoretically, it should always work as long as you have this line:

> %CARLA_ROOT%\PythonAPI\carla

Lu7913, 2020-07-23:
Hi, I set the path accordingly to manuals which is 

> PYTHONPATH=%PYTHONPATH%:%CARLA_ROOT%\PythonAPI (C:\Users\%user%\AppData\Local\Programs\Python\Python37:D:\%user%\carla\PythonAPI\carla)

with %PYTHONPATH% before %CARLA_ROOT%, removing %PYTHONPATH%, helped.

Now when running `python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld` and then `python manula.control.py` I got:
>No module named 'examples.manual_control'

Are there more Paths to be set different than in manual?
 
To add why I had to downgrade tensorflow, because tensorflow.contrib is apparently not supported in the tensorflow>2.xx. So running with tensorflow 1.15 resulted in:
`The TensorFlow Distributions library has moved to TensorFlow Probability (https://github.com/tensorflow/probability). You should update all references to use `tfp.distributions` instead of `tf.distributions`.`
despite I did the necessary updates, problem remained until now when I have fixed the path. 
Should ScenarioRunner work with tensorflow greater than 2.00? Because downgrading it created a lot of mess with different  modules compatibility. Or it was just effect of false PYTHONPATH?

Thank you for your help.

glopezdiest, 2020-07-23:
Well, you solved one problem, but you definitely need more variables to be added. You seem to also be missing the PythonAPI:

`set PYTHONPATH=%PYTHONPATH%:%CARLA_ROOT%\PythonAPI`

From the Docs, you need to add all of this to the PYTHONPATH for ScenarioRunner to work. Please make sure you have them all

[image]

Lu7913, 2020-07-23:
Yes of course, I had all this paths added as described in docs, days ago.  The thing just don't work for me. Could this be related to having Carla installed on different disk that python? 
I tried for all other paths the same thing, that helped me minutes for one, but it seems that it is not working.

glopezdiest, 2020-07-23:
By using the set command, you are only adding them to the current terminal, but whenever you close it, the PYTHONPATH is gone.

You have to permanently set them, by going to *edit the environment variables of this account*. This can be quickly accessed by writing *env* on the Windows' search panel.

From the PYTHONPATH you posted, you seem to lack many of the lines

Lu7913, 2020-07-23:
Yes I did all that. 
[image]

glopezdiest, 2020-07-23:
I see, I misunderstood you then. I actually don't know why it is failling then.

If you can run the FollowLeadingVehicle scenario without problems, then the *.egg* file being found, and also the *PythonAPI\carla* folder, but for some reason you can't find the PythonAPI

Lu7913, 2020-07-28:
Hi,
I managed to avoid problems with path, by not building Carla but, downloading and extracting it.
When trying to run example scenario with ScenarioRunner i got the following error:
`AttributeError: module 'signal' has no attribute 'SIGHUP'`
Apparently this should not be run on windows. I commented out line 128 in scenario_runner.py and for now, seems to work. how to avoid this problem on windows otherwise?

There remain a problem with opening Unreal Engine project, when trying to open it and rebuilding it crashes and i got an error:
`CarlaUE4 could not be compiled. Try rebuilding from source manually`, because project was made in different version, but I use UE 4.24.3.

But if I understand correctly, now i can't rebuild it from source, because I downloaded it and extracted? Correct me if I am wrong.
How 
As it seems when I build Carla from source, it is not building correctly, what could be the issue?
I tried building from source many times, with different pythons versions, assets, but only the latest Carla.

glopezdiest, 2020-07-28:
> AttributeError: module 'signal' has no attribute 'SIGHUP'

Yeah, we fixed that [here](https://github.com/carla-simulator/scenario_runner/pull/561) but i don't think it is in any release yet. As far as I know,  the signal just doesn't exist for Windows.

Regarding your building problems, I actually don't know, I had no issues installing CARLA. If I understand correctly, you can *make clean* , remove the carla folder and it should remove everything, but I'm not sure.

Lu7913, 2020-07-28:
I did that when building from source, now I can't build UE project manually because I just downloaded and extracted Carla, right?

I'm planing to switch to Linux, but for now I'm stuck with Windows.

Again, thank you.

Lu7913, 2020-08-24:
Hi,
after some time me again. Yeah so... prebuild Carla works on Windows, I have built Carla from source on Kubuntu, which was working great with ScenarioRunner until I tried to install ROS and changed default python. I switched back default python and there is same problem that I have on Windows. Clearly I'm doing something wrong, but don't know what.

If this maybe is pointing to where the problem might be? I will also mention that when building on Windows .egg for default python is created only, on Kubuntu .egg files were created for python2 and python3.

I'm doing quite a lot of research, but no luck. To me this seems to be some silly mistake which I'm repeating.

Lu7913, 2020-08-26:
I have managed to solve the issue on Kubuntu, with manually deleting Carla module and then reinstalling it.
With Windows there seems to be only one specific way of setting up PYTHONPATH, which works for me:

%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla\dist\carla-<VERSION>.egg
%CARLA_ROOT%\PythonAPI\carla\agents
%CARLA_ROOT%\PythonAPI\carla
%CARLA_ROOT%\PythonAPI

I have talked to some people about this, but they are also not sure why this works and nothing else.

closed 2020-08-26 by Lu7913

aczy118, 2020-09-26:
Hi,
Can you please explain how did you solve the problem for Windows? I have tried your last entry but I am still getting 'no module named carla' . Why is it not working? 
Thanks.

Lu7913, 2020-09-28:
First make sure that Carla server is running fine and you can connect python client to it. Than make sure that you replace text between %% with your corresponding location.  
[image]

Try with prebuild version of Carla first. I haven't had issues with that.

sterres, 2021-06-11:
Your problem might be the spelling error in the last line of the command list - the use ":" as separator instead of ";".
It is fixed in the documenation but the screenshot here is still outdated.

sterres, 2021-06-11:
> Hi, I set the path accordingly to manuals which is
> 
> > PYTHONPATH=%PYTHONPATH%:%CARLA_ROOT%\PythonAPI (C:\Users%user%\AppData\Local\Programs\Python\Python37:D:%user%\carla\PythonAPI\carla)
> 
> with %PYTHONPATH% before %CARLA_ROOT%, removing %PYTHONPATH%, helped.
> 
> Now when running `python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld` and then `python manula.control.py` I got:
> 
> > No module named 'examples.manual_control'
> 
> Also no actors are spawned.
> 
> Are there more Paths to be set different than in manual?
> 
> To add why I had to downgrade tensorflow, because tensorflow.contrib is apparently not supported in the tensorflow>2.xx. So running with tensorflow 1.15 resulted in:
> `The TensorFlow Distributions library has moved to TensorFlow Probability (https://github.com/tensorflow/probability). You should update all references to use `tfp.distributions`instead of`tf.distributions`.`
> despite I did the necessary updates, problem remained until now when I have fixed the path.
> Should ScenarioRunner work with tensorflow greater than 2.00? Because downgrading it created a lot of mess with different modules compatibility. Or it was just effect of false PYTHONPATH?
> 
> Thank you for your help.

Just came accross the same isseue .... my solution: 

Go to your Carla examples folder (subfolder PythonAPI\examples) and create a file with name **"__init__.py"**. This will allow the script to load the Carla Python modules without error