carla-simulator/scenario_runner #600: The scenario cannot be loaded. frame == _episode->GetState()->GetTimestamp().frame Error

LucyWang01, 2020-07-22:
**Describe our question or idea**
When I run certain scenarios using scenario runner 0.9.6 (e.g. ControlLoss, DynamicObjectCrossing, etc.), I get either of the following errors:

"Preparing scenario: DynamicObjectCrossing_5
The scenario cannot be loaded
frame == _episode->GetState()->GetTimestamp().frame
No more scenarios .... Exiting"

"Preparing scenario: DynamicObjectCrossing_5
The scenario cannot be loaded
Error: Unable to spawn vehicle vehicle.lincoln.mkz2017 at Transform(Location(x=151.800003, y=17.700001, z=10.510000), Rotation(pitch=0.000000, yaw=180.000000, roll=0.000000))
No more scenarios .... Exiting"

"Traceback (most recent call last):
  File "scenario_runner.py", line 424, in <module>
    SCENARIORUNNER.run(ARGUMENTS)
  File "scenario_runner.py", line 293, in run
    if not self.load_world(args, config.town):
  File "scenario_runner.py", line 243, in load_world
    self.world.tick()
RuntimeError: frame == _episode->GetState()->GetTimestamp().frame"

However, it loads the scenario successfully on occasion (around 5% of the times). I'm not sure why I'm getting these different errors when I'm using the same commands every time (python3 scenario_runner.py --scenario DynamicObjectCrossing_5 --reloadWorld). 

When I ran other scenarios (e.g. FollowLeadingVehicles), the scenario successfully loads almost every time. 

Any idea on why I'm getting these errors when I'm running certain scenarios?

**Expected behavior**
Scenarios successfully loaded. No error occurs.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version: CARLA 0.9.6
 - Python version [e.g. 3.5]
 - Scenario Runner Version: 0.9.6

glopezdiest, 2020-07-24:
Hey @KeyingLucyWang.

Regarding your spawn errors, that is due to the ego vehicle trying to spawn at an already occupied space. I think we fixed that by adding a default vertical shift to avoid collisions with the ground.

Regarding the frame error, at that time the ticking structure of ScenarioRunner was outdated and failed, which why sometimes the world.tick caused errors.

I'd recommend upgrading to 0.9.7 if you can, which is a much more up to date version.

KeyingLucyWang, 2020-07-24:
Thank you! I updated to 0.9.9 and stopped getting the above errors. 

However, I'm consistently getting a bind error when I run manual_control.py with the scenario runner. 

In one terminal, I ran scenario_runner.py: 
$ python3 scenario_runner.py --scenario FollowLeadingVehicle_5
Preparing scenario: FollowLeadingVehicle_5
ScenarioManager: Running scenario FollowVehicle

In another terminal:
$ python3 manual_control.py
ERROR: trying to create rpc server for traffic manager; but the system failed to create because of bind error.
Traceback (most recent call last):
  File "manual_control.py", line 681, in main
    game_loop(args)
  File "manual_control.py", line 615, in game_loop
    controller = KeyboardControl(world, args.autopilot)
  File "manual_control.py", line 190, in __init__
    world.vehicle.set_autopilot(self._autopilot_enabled)
RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.

Any idea on what might be causing this error?

glopezdiest, 2020-07-27:
Yeah, that's because of the TrafficManager. This generally happens because of two reasons:
- There is another instance of ScenarioRunner running (which wasn't closed properly)
- You launched the manual control too soon, before the scenario was running, and it has problems connecting to the ego vehicle

Did you do the second option? Also, what version of CARLA are you using? I'd recommend upgrading to 0.9.9.4, as 0.9.9 has an error related to world loading, and specially affects ScenarioRunner.

closed 2020-08-03 by glopezdiest