carla-simulator/scenario_runner #601: ScenarioRunner AttributeError: module 'signal' has no attribute 'SIGHUP'

hnuma, 2020-07-22:
Hello,

I want to ask an error I take on installation steps of ScenarioRunner. When I run "python scenario_runner.py --scenario OppositeVehicleRunningRedLight011" command, I encounter AttributeError: module 'signal' has no attribute 'SIGHUP'.

How can I remove this error, could you help?

[image]

 - OS: [Windows 10]
 - CARLA 0.9.9.4
 - Python 3.7
 - [scenario_runner-0.9.9]

DevotionGit, 2020-07-23:
Windows  not support  this.comment this line,it run well.

hnuma, 2020-07-23:
@DevotionGit thank you for your reply. I did it, it worked.

closed 2020-07-23 by hnuma