carla-simulator/scenario_runner #602: Error using get_light_state

s-rajab, 2020-07-23:
Hello,

I am trying to get the light status of all vehicles in the world:

`import carla
from srunner.scenariomanager.carla_data_provider import CarlaDataProvider, CarlaActorPool
world = CarlaDataProvider.get_world()
actor_list = world.get_actors()
for target_vehicle in self.actor_list.filter('vehicle.*'):
      lights = target_vehicle.get_light_state()`

However, I am getting the following error:

AttributeError: 'Vehicle' object has no attribute 'get_light_state'
'Vehicle' object has no attribute 'get_light_state'

[image]

I appreciate any help.

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - CARLA Version 0.9.7
 - Python version 3.5
 - Version 0.9.7

glopezdiest, 2020-07-24:
Hey @s-rajab. Vehicle lights were introduced in CARLA 0.9.8, so you are just one version short.

s-rajab, 2020-07-24:
Thanks @glopezdiest. I guess it's time to upgrade then.

closed 2020-07-24 by s-rajab