carla-simulator/scenario_runner #610: Vehicle max acceleration

lucasnmatias, 2020-08-06:
**Is your feature request related to a problem? Please describe.**
Hi, I work at IVEX and our focus is on the evaluation of autonomous vehicles behavior and decision making. We use CARLA to implement challenging scenarios to evaluate vehicles behavior in those scenarios. Right we are struggling to control the vehicles acceleration.

**Describe the solution you'd like**
Here is my question, is there any easy way to limit the max acceleration of a vehicle in CARLA? We want the behavior of vehicle to be normal, but limit the max acceleration to know pass a specific value.

**Describe alternatives you've considered**
I have seen that in ROS Bridge there is the ackermann commands but we use the OpenSCENARIO to control the vehicles, so I would like to know if there is a simpler solution inside scenario_runner or inside OpenSCENARIO to limit the vehicle max acceleration.

fabianoboril, 2020-08-10:
Hi Lucas,
thank you for opening this issue. In fact, for OpenSCENARIO you specify for each vehicle a maximum acceleration value:
https://github.com/carla-simulator/scenario_runner/blob/585a3b912abf6550a8a4defa19f84f016a1192da/srunner/examples/FollowLeadingVehicle.xosc#L16
Another option is to provide an additional property for a vehicle controller, to limit the acceleration:
https://github.com/carla-simulator/scenario_runner/blob/585a3b912abf6550a8a4defa19f84f016a1192da/srunner/examples/FollowLeadingVehicle.xosc#L79

However, both options do not yet work out of the box.

If you are interested in contributing to scenario, we could outline required changes, to enable you to get this to work on your side asap.

lucasnmatias, 2020-08-10:
That would be great, if you can address the required changes I could study and check if I could implement and contribute to it!