carla-simulator/scenario_runner #616: Ego vehicle loses control after setting light status

s-rajab, 2020-08-14:
I am adding a WaypointFollower behavior for the ego vehicle to a parallel root in my scenario.

I have a light_sequence for changing  the ego vehicle light status that includes:
- checking if distance between ego and other vehicle is below a threshold using: InTriggerDistanceToVehicle
- Then, change the ego vehicle lights status using a new atomic behavior that I have defined

I add the light_sequence as a child to the parallel root.

However, as soon as the ego lights status change, the ego vehicle WaypointFollower disengages. It seems that the vehicle.set_light_state is causing the WaypointFollower to loose handle on the ego vehicle. Although, it is expected that the WaypointFollower should keep driving the ego vehicle since it is a parallel behavior to the light_sequence.

I appreciate any help on this.

Desktop :
 - OS: Ubuntu 16.04
 - CARLA Version: 0.9.9.4
 - Python version: 3.7
 - Version: master

fabianoboril, 2020-08-27:
Are you usng OpenSCENARIO or py_trees directly?

s-rajab, 2020-08-28:
Hi @fabianoboril ,

Thank you for the --debug hint. It is very useful.

Here is my _create_behavior class:
`# Source and Sink
        actor_source = ActorSource(['vehicle.tesla.model3'], self._source_transform, self._source_gap, self._blackboard_queue_name)
        actor_sink = ActorSink(self._sink_location, threshold=0) 

        # Other vehicle driving
        other_driving = WaypointFollower(self.other_actors[0], self._opposite_speed,
                                         blackboard_queue_name=self._blackboard_queue_name, avoid_collision=True, sampling_radius=0.5, name="OtherDriving") 

        # Ego parallel
        ego_parallel = py_trees.composites.Parallel(policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ALL,"ego_behavior")
        
        #Ego driving
        ego_parallel.add_child(WaypointFollower(self.ego_vehicles[0], self._HV_speed, sampling_radius=0.5,name="EgoDriving"))
        
        # Lights control at a particular point
        lights_sequence = py_trees.composites.Sequence("ChangingLights")
        lights_sequence.add_child(InTriggerDistanceToVehicle(self.ego_vehicles[0], self.other_actors[0], self.Trigger_Distance, name="DistanceToOther"))
        light_change=carla.VehicleLightState(carla.VehicleLightState.LeftBlinker | carla.VehicleLightState.RightBlinker | carla.VehicleLightState.Position)
        lights_sequence.add_child(VehicleLightsControls(actor=self.ego_vehicles[0], light_status=light_change, name="LightsControl"))
        lights_sequence.add_child(TimeOut(60))
        ego_parallel.add_child(lights_sequence)
        
        # End condition
        end_condition = DriveDistance(self.ego_vehicles[0], self._ego_vehicle_drive_distance, name="EndCondition")

        # Root
        parallel_root = py_trees.composites.Parallel(policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE, name="Root")
        parallel_root.add_child(actor_source)
        parallel_root.add_child(actor_sink)
        parallel_root.add_child(other_driving)
        parallel_root.add_child(ego_parallel)
        parallel_root.add_child(end_condition)
        scenario_sequence = py_trees.composites.Sequence()
        scenario_sequence.add_child(ActorTransformSetter(self.other_actors[0], self._second_actor_transform))
        scenario_sequence.add_child(parallel_root)

        return scenario_sequence`

This new lights control behavior I created:
`class VehicleLightsControls(AtomicBehavior):    

    """
    A class to set lights status on a vehicle

    Important parameters:
    - name: Name of the atomic behavior
    - actor: the vehicle that we are changing the light_status for
    - light_status: the new light state to be set
    """

    def __init__(self, actor, light_status=carla.VehicleLightState(carla.VehicleLightState.NONE), name="VehicleLights"):
        """
        Default init. Has to be called via super from derived class
        """
        super(VehicleLightsControls, self).__init__(name)
        self.logger.debug("%s.__init__()" % (self.__class__.__name__))
        self.name = name
        self._actor = actor
        self.light_status = light_status

    def update(self):
        self._actor.set_light_state(self.light_status)
        new_status = py_trees.common.Status.SUCCESS
        return new_status`

I took a snap shot from the behavior tree state when the vehicle lost control. Based on my understanding when looking at the tree, the ego vehicle EgoDriving  WaypointFollower returns SUCCESS after the lights were changes. However, this should not be the case as WaypointFollower documentation mentions that: "If no plan is provided, the actor will follow its foward waypoints indefinetely."

[image]

Thanks!

fabianoboril, 2020-09-02:
Hi,
from the code it looks ok. However, the debug-info of the tree reveals something interesting. You Waypoint-Following behavior is already completed. This would explain, why there is a control loss. Can you check, what causes the behavior to end with success?
Therefore, please go to:
https://github.com/carla-simulator/scenario_runner/blob/76d56c615c6af9e63f53bd60978f2be56d8a2883/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py#L1502
In here, you find a couple of locations, where the behavior status changes to success. Would it be possible for you, to check which one is triggering your behavior to end? This will help us to find the root cause of your issue.

s-rajab, 2020-09-02:
Hi @fabianoboril,

The second one is triggering the behavior end. It seems that after the VehicleLightState is set on the ego vehicle, "Resetting ego-vehicle!" is printed on the terminal and self._local_planner_dict is becoming empty at that point. Thus,  `update(self)` does not go in the ` for actor in self._local_planner_dict: ` loop and the scenario ends in success. Not sure why this is happening though. Look like it is related to setting the VehicleLightState.

fabianoboril, 2020-09-03:
Can you check if Line 1540 is executed once the LightSetter is called?
https://github.com/carla-simulator/scenario_runner/blob/76d56c615c6af9e63f53bd60978f2be56d8a2883/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py#L1540

s-rajab, 2020-09-03:
Line 1540 is no longer executed after the LightSetter is called.

fabianoboril, 2020-09-04:
Ah, I think I found the issue. 
As you haven't defined an initialise() method for your LightSetter, the parent method is called, which terminates the WaypointFollower. Please define an initalise() method in your LightSetter and do not call the parent method. Something like:

```
    def initialise(self):
        return
```

Does it help?

s-rajab, 2020-09-08:
That solved the issue. Thank you!

closed 2020-09-08 by fabianoboril