carla-simulator/scenario_runner #621: The scenario I created is not supported

Francescovar, 2020-08-29:
Hi everyone,

I was trying to build Highway merging scenario in Carla as shown in the video on Introduction to Scenario runner- https://www.youtube.com/watch?v=dcnnNJowqzM&t=932s. But I receive this error:

```
Preparing scenario: slowtraffic
Scenario 'slowtraffic' not supported ... Exiting
Destroying ego vehicle 206
```
I set the environment as suggested in docs, then I created the script of SlowTraffic and the xml file as well.
I followed all the instructions provided by the Docs for creating a new scenario in scenario_runner. 
Can someone help me? Thank you

 - OS: [Windows 10]
 - CARLA Version [e.g. CARLA 0.9.9]
 - Python version [e.g. 3.7]

glopezdiest, 2020-08-31:
Hey @Francescovar. Where are you saving the *.py* file. By default, SR looks at all the files inside *srunner/scenarios*. Additionally, if you don't want your file to be located there, you can use the *--additionalScenario* argument, which should be the path locating the file.

Francescovar, 2020-08-31:
Hi @glopezdiest thank you for your reply!

My .py file is in _srunner/scenarios_, as the .xml is in _srunner/examples_ 
My scenario appears in the list of scenarios when I launch "--list" command, so I tried the following:

`py -3.7 scenario_runner.py --scenario slowtraffic`
`py -3.7 scenario_runner.py --scenario slowtraffic --additionalScenario path_to_.py_file`
`py -3.7 scenario_runner.py --scenario slowtraffic --additionalScenario path_to_.py_file --configFile path_to_.xml_file`

But none of them works, it always gives me the same error. I don't know, there is something to do before creating a new scenario? Any other idea? Thank you very much

Francescovar, 2020-09-01:
Ok, it seems I solved the problem just by changing "slowtraffic" in "SlowTraffic" in xml file, because the class is defined with capital letters. So now that error doesn't appear anymore, but now I have the following error:

```
Preparing scenario: SlowTraffic
The scenario cannot be loaded
Traceback (most recent call last):
  File "scenario_runner.py", line 355, in _load_and_run_scenario
    self._args.debug)
  File "C:\Users\franc\Desktop\CARLA\WindowsNoEditor\PythonAPI\examples\scenario_runner-0.9.9\srunner\scenarios\slow_traffic.py", line 34, in __init__
    super(SlowTraffic, self).__init__("SlowTraffic", ego_vehicles, config,world, debug_mode, criteria_enable=criteria_enable)
  File "C:\Users\franc\Desktop\CARLA\WindowsNoEditor\PythonAPI\examples\scenario_runner-0.9.9\srunner\scenarios\basic_scenario.py", line 49, in __init__
    self._initialize_actors(config)
  File "C:\Users\franc\Desktop\CARLA\WindowsNoEditor\PythonAPI\examples\scenario_runner-0.9.9\srunner\scenarios\slow_traffic.py", line 65, in _initialize_actors
    new_actors = CarlaActorPool.request_new_batch_actors("vehicle.*",len(spawn_transforms),spawn_transforms)
  File "C:\Users\franc\Desktop\CARLA\WindowsNoEditor\PythonAPI\examples\scenario_runner-0.9.9\srunner\scenariomanager\carla_data_provider.py", line 697, in request_new_batch_actors
    actors = CarlaActorPool.setup_batch_actors(model, amount, spawn_points, hero, autopilot, random_location)
  File "C:\Users\franc\Desktop\CARLA\WindowsNoEditor\PythonAPI\examples\scenario_runner-0.9.9\srunner\scenariomanager\carla_data_provider.py", line 685, in setup_batch_actors
    batch.append(SpawnActor(blueprint, spawn_point).then(SetAutopilot(FutureActor, autopilot)))
Boost.Python.ArgumentError: Python argument types in
    SpawnActor.__init__(SpawnActor, ActorBlueprint, Waypoint)
did not match C++ signature:
    __init__(struct _object * __ptr64, struct carla::rpc::Command::SpawnActor)
    __init__(class boost::python::api::object, class carla::client::ActorBlueprint blueprint, class carla::geom::Transform transform, class boost::shared_ptr<class carla::client::Actor> parent)
    __init__(class boost::python::api::object, class carla::client::ActorBlueprint blueprint, class carla::geom::Transform transform, unsigned int parent_id)
    __init__(class boost::python::api::object, class carla::client::ActorBlueprint blueprint, class carla::geom::Transform transform)
    __init__(struct _object * __ptr64)
Python argument types in
    SpawnActor.__init__(SpawnActor, ActorBlueprint, Waypoint)
did not match C++ signature:
    __init__(struct _object * __ptr64, struct carla::rpc::Command::SpawnActor)
    __init__(class boost::python::api::object, class carla::client::ActorBlueprint blueprint, class carla::geom::Transform transform, class boost::shared_ptr<class carla::client::Actor> parent)
    __init__(class boost::python::api::object, class carla::client::ActorBlueprint blueprint, class carla::geom::Transform transform, unsigned int parent_id)
    __init__(class boost::python::api::object, class carla::client::ActorBlueprint blueprint, class carla::geom::Transform transform)
    __init__(struct _object * __ptr64)
Destroying ego vehicle 327
No more scenarios .... Exiting
```

Any idea on how to resolve?

shimizhang, 2021-01-28:
Hi, 

I am testing this SlowTraffic scenario TOO. Did you solve the problem now? If solved, hope you can kindly provide the solution. 

Best

bigdayangyu, 2021-04-15:
Hi @Francescovar and @coolCoco-MiMi, the example used wrong data type for `SpawnActor()` method. Instead of using` carla.waypoint`, you should use `carla.Transform` as the correct data type for `SpawnActor()` method.  I solved this problem by changing the `spawn_transforms.append(spawn_waypoint)` to `spawn_transforms.append(spawn_waypoint.transform)` and that solved the problem. 

I hope this helps, cheers!

closed 2021-11-23 by glopezdiest