carla-simulator/scenario_runner #625: Bug? OpenSCENARIO: UnboundLocalError: local variable 'rel_pos' referenced before assignment

r-snijders, 2020-09-01:
Hello, is this a bug? I hope my report helps, thanks.

**Describe the bug**
In the current master branch (commit 33dbed109da6a12c82df4f3388f6c21dbd4325dc) I run into the following problem:

```bash
Traceback (most recent call last):
  File "scenario_runner.py", line 543, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 533, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 445, in run
    result = self._run_openscenario()
  File "scenario_runner.py", line 433, in _run_openscenario
    config = OpenScenarioConfiguration(self._args.openscenario, self.client)
  File "/home/carla/simulation/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 59, in __init__
    self._parse_openscenario_configuration()
  File "/home/carla/simulation/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 91, in _parse_openscenario_configuration
    self._set_actor_information()
  File "/home/carla/simulation/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 243, in _set_actor_information
    actor.transform = self._get_actor_transform(actor.rolename)
  File "/home/carla/simulation/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 330, in _get_actor_transform
    position, actor_list=self.other_actors + self.ego_vehicles)
  File "/home/carla/simulation/scenario_runner/srunner/tools/openscenario_parser.py", line 561, in convert_position_to_transform
    orientation = rel_pos.find('Orientation')
UnboundLocalError: local variable 'rel_pos' referenced before assignment
```
This seems like a bug to me. Should `rel_pos` not be `lane_pos`?
The problem resolves when I change it.

**To Reproduce**
I could provide the OpenSCENARIO file used, but then I need to verify/modify it first due to possible confidentiality issues.

**Desktop:**
 - OS: latest carlasim docker image (0.9.9): https://hub.docker.com/r/carlasim/carla
 - CARLA Version 0.9.9
 - Python version 3.7 (manually installed using deadsnakes/ppa)
 - Version 0.9.9 (latest, same as provided in docker image), installed egg using `python3.7 -m easy_install PythonAPI/carla/dist/carla-0.9.9-py3.7-linux-x86_64.egg`.

fabianoboril, 2020-09-02:
Hi,
I took a look into the code and couldn't find the problem straight away. The code in master at the reported line 561 looks different https://github.com/carla-simulator/scenario_runner/blob/76d56c615c6af9e63f53bd60978f2be56d8a2883/srunner/tools/openscenario_parser.py#L561
and rel_pos is assigned in line 437ff
https://github.com/carla-simulator/scenario_runner/blob/76d56c615c6af9e63f53bd60978f2be56d8a2883/srunner/tools/openscenario_parser.py#L437
Could you check if your code is up-to-date, or if you have any modifications?

r-snijders, 2020-09-02:
Yeah sorry, I noticed that I didn't use the latest commit. I did check it with the latest commit in master and changed the commit id above, but I forgot to modify the traceback as well.

So, the problem still exists in the latest commit but the traceback above is incorrect, I'll change it later today once I have access to the other machine again.

r-snijders, 2020-09-03:
I've updated the traceback in the first comment.
I am using commit 76d56c615c6af9e63f53bd60978f2be56d8a2883 from the master branch. asdfasdf

fabianoboril, 2020-09-03:
Ah, found it. Yes, it has to be lane_pos not rel_pos. Added PR #627

closed 2020-09-04 by glopezdiest

markfilan, 2021-12-28:
An UnboundLocalError is raised when a local variable is referenced before it has been assigned. In most cases this will occur when trying to modify a local variable before it is actually assigned within the local scope. Python doesn't have variable declarations, so it has to figure out the scope of variables itself. It does so by a simple rule: If there is an assignment to a variable inside a function, that variable is considered local. 

Python has lexical scoping by default, which means that although an enclosed scope can access values in its enclosing scope, it cannot modify them (unless they're declared global with the global keyword). A closure binds values in the enclosing environment to names in the local environment. The local environment can then use the bound value, and even reassign that name to something else, but it can't modify the binding in the enclosing environment. UnboundLocalError happend because when python sees an assignment inside a function then it considers that variable as local variable and will not fetch its value from enclosing or global scope when we execute the function. However, to modify a [global variable](http://net-informations.com/python/iq/global.htm) inside a function, you must use the global keyword.

PR #627 (2020-09-04): Fixed OSC LanePosition handling
Fixes #625

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/627)