carla-simulator/scenario_runner #626: OpenSCENARIO support - PrivateActions

glopezdiest, 2020-09-02:
This issue is meant to track the support of PrivateActions within OSC 1.0:

- [x] ActivateControllerAction          Activates CARLA's autopilot
- [x] ControllerAction                      
- [x] LaneChangeAction                  Some dynamics are still ignored
- [ ] LaneOffsetAction
- [ ] LateralDistanceAction
- [ ] LongitudinalDistanceAction
- [x] SpeedAction
- [ ] SynchronizeAction
- [x] TeleportAction
- [ ] VisibilityAction
- [ ] AcquirePositionAction
- [x] AssignRouteAction
- [ ] FollowTrajectoryAction

r-snijders, 2020-09-03:
Thanks for working on improving support for OSC1.0! :+1: 
I am especially interested in support for the LaneOffsetAction.

glopezdiest, 2020-09-09:
As of #628, AcquirePositionAction is now supported at the *Story* part

glopezdiest, 2020-12-03:
As of #689, SynchronizeAction is now supported