carla-simulator/scenario_runner #632: RelativeTargetSpeed OSC issue

sagar-g-v, 2020-09-14:
I was trying to use relative speed in open scenario, but i got the below error.

Traceback (most recent call last):
  File "scenario_runner.py", line 367, in _load_and_run_scenario
    timeout=100000)
  File "/home/harsha/scenario_runner/srunner/scenarios/open_scenario.py", line 188, in __init__
    terminate_on_failure=False, criteria_enable=criteria_enable)
  File "/home/harsha/scenario_runner/srunner/scenarios/basic_scenario.py", line 63, in __init__
    behavior = self._create_behavior()
  File "/home/harsha/scenario_runner/srunner/scenarios/open_scenario.py", line 310, in _create_behavior
    child, joint_actor_list[actor_id], self.config.catalogs)
  File "/home/harsha/scenario_runner/srunner/tools/openscenario_parser.py", line 983, in convert_maneuver_to_atomic
    target_speed,
UnboundLocalError: local variable 'target_speed' referenced before assignment
local variable 'target_speed' referenced before assignment

I was able to fix it from my end, but not sure if it is up to open-scenario standards.

**Desktop:**
 - OS: 18.04
 - CARLA Version: CARLA 0.9.9.4
 - Python version : 2.7
 - Version: 0.9.9

ThrashAbaddon, 2020-09-23:
@sagar-g-v Could you please submit a PR with what you have?

fabianoboril, 2020-09-24:
This seems to be included in #638

sagar-g-v, 2020-09-24:
Yes. I will submit a pull request for this.

PR #639 (2020-09-25): Fixed OSC RelativeTargetSpeed error

#### Description
For RelativeTargetSpeed `target_speed` value was not initialised as it was not possible to find this value before the event is triggered. so the `target_speed` is initialised to 0.0 when parsed.

When the ChangeActorTargetSpeed is initialised `target_speed` is set to reference actor speed.

Fixes   #632

#### Where has this been tested?

  * **Platform(s):**Ubuntu 18.04
  * **Python version(s):**  2.7
  * **Unreal Engine version(s):** 4.24
  * **CARLA version:**  0.9.9.4 & 0.9.10

#### Possible Drawbacks

None

---
This change is [[image]]( [... cut]
closed 2020-09-25 by fabianoboril