carla-simulator/scenario_runner #633: jenkins file

soheilade, 2020-09-15:
what is 
`jenkinsLib = load("/home/jenkins/scenario_runner.groovy") `
in `jenkinsfile `for? where is `scenario_runner.groovy` supposed to be located?

fabianoboril, 2020-09-15:
This is our internal jenkins setup. I guess you can ignore this file, as it is not relevant for the normal use of ScenarioRunner. @melviso-osvf: Anything to add?

melviso-osvf, 2020-09-16:
Ok. That file is more or less the same used in carla jenkins (I have modified it a little for the squeduling of carla instances). This file is located inside the AWS Jenkins server, where that script would be executed.

soheilade, 2020-09-16:
thanks guys for your comments, so i assume if i remove the lines related to jenkinsLib, it should run without problem, is the docker image of scenario runner available somewhere like in docker hub?

soheilade, 2020-09-17:
@fabianoboril , I tried to remove the file and while running this command:
 `run apt-get update && apt-get install --no-install-recommends -y libpng16-16=1.6.34-1ubuntu0.18.04.2 libtiff5=4.0.9-5ubuntu0.3 libjpeg8=8c-2ubuntu8 build-essential=12.4ubuntu1 wget=1.19.4-1ubuntu2.2 git=1:2.17.1-1ubuntu0.7  python3.6=3.6.9-1~18.04ubuntu1 python3.6-dev=3.6.9-1~18.04ubuntu1 python3-pip=9.0.1-2.3~ubuntu1.18.04.1  && rm -rf /var/lib/apt/lists/*`
I get this error:
```
[91mE: Version '3.6.9-1~18.04ubuntu1' for 'python3.6' was not found
E: Version '3.6.9-1~18.04ubuntu1' for 'python3.6-dev' was not found
E: Version '9.0.1-2.3~ubuntu1.18.04.1' for 'python3-pip' was not found
[0mThe command '/bin/sh -c apt-get update && apt-get install --no-install-recommends -y libpng16-16=1.6.34-1ubuntu0.18.04.2 libtiff5=4.0.9-5ubuntu0.3 libjpeg8=8c-2ubuntu8 build-essential=12.4ubuntu1 wget=1.19.4-1ubuntu2.2 git=1:2.17.1-1ubuntu0.7  python3.6=3.6.9-1~18.04ubuntu1 python3.6-dev=3.6.9-1~18.04ubuntu1 python3-pip=9.0.1-2.3~ubuntu1.18.04.1  && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
```

any idea?

fabianoboril, 2020-09-17:
There is no public docker image for ScenarioRunner. You can simply download it, and run it with Python.

melviso-osvf, 2020-09-18:
Also, I think you could use the current dockerfile for building the image locally, if you want to do that.

soheilade, 2020-09-30:
thanks guys!

closed 2020-09-30 by soheilade