carla-simulator/scenario_runner #634: Change viewpoint

TKeutgens, 2020-09-16:
I was finaly able to play my own XOSC scenario with the scenario_runner.

What is the best way to set the specator view to the ego vehicle or to a specific position? Are there examples or tutorials to change the view with scenario runner?

 - OS: Windows10
 - CARLA Version: CARLA 0.9.10
 - Python version: 3.7
 - Scenario Runner: 0.9.9

sagar-g-v, 2020-09-21:
Hi,

I think there is a way to set spectator view to follow ego.

the following chages needs to be made in `manual_control.py`
```python
class World(object):
    def __init__(self, carla_world, hud, args):
        .
        .
        self.spectator = self.world.get_spectator()
        self.spectator_transform = self.spectator.get_transform()
        .
        .

    def tick(self, clock):
        .
        .
        transform = self.player.get_transform()
        location = carla.Location(x=transform.location.x, y=transform.location.y, z=70)
        self.spectator.set_transform(
            carla.Transform(location, carla.Rotation(yaw=180.0, pitch=-90.0)))
        .
        .
```

try this and let me know.

glopezdiest, 2021-11-23:
Closing this issue as another user has already answered the problem. Feel free to open it if that isn't the case

closed 2021-11-23 by glopezdiest