carla-simulator/scenario_runner #636: Contributing with Scenarios (OpenScenario)

lucasnmatias, 2020-09-18:
**Is your feature request related to a problem? Please describe.**
Hi, I work at IVEX and recently we have worked with scenario_runner to build scenarios and make the safety assessment of the vehicles behaviors. To do so we have implemented some of the NHTSA Scenarios from CARLA challenge on OpenSCENARIO and we would like to contribute to CARLA making a PR with those OpenSCENARIO scripts of NHTSA Scenarios. So I would like to know how could we make a PR of it?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

fabianoboril, 2020-09-19:
Hi,
this contribution sounds amazing and would be a great benefit for us and the community :)
Regarding the PR: There are several options but the easiest is to fork the scenario_runner repo, commit your changes on a local branch in the fork, and then do a PR from that branch back to ScenarioRunner.
https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

Would this work for you?

jimmyw404, 2020-10-09:
@lucasnmatias  Would be interested in seeing this and would also be interested in hearing your process for creating these scenarios!

lucasnmatias, 2020-10-09:
@fabianoboril yes it would work! Sorry for the delay, within some days I will make the PR.
@jimmyw404 yes, we can discuss and I share the experience of the scenario creation process.

ajdroid, 2021-02-08:
Hi @lucasnmatias, is this a PR I can find somewhere already? Thanks! :)