carla-simulator/scenario_runner #643: A possible bug rotate_point() for RunningRedLightTest

emailweixu, 2020-09-29:
**Describe the bug**

It seems there is bug of rotate_point() function for  RunningRedLightTest

https://github.com/carla-simulator/scenario_runner/blob/e4241813f39554be048c274c8e5e05f778d85f6a/srunner/scenariomanager/scenarioatomics/atomic_criteria.py#L1829-L1835

If the angle is 0, the returned vector becomes Vector3D(point.x, -point.y, point.z), which is different from the original vector. And function doesn't even perform a rotation. Because the transformation matrix it uses is [[cos, -sin], [sin, -cos]], which is not a rotation.

fabianoboril, 2020-09-29:
@glopezdiest Can you have a look into this?

glopezdiest, 2020-09-29:
Sure, I'll take a look and report back

glopezdiest, 2020-09-30:
Hey @emailweixu, many thanks for seeing that. Indeed, that rotation "matrix" was wrong, and vectors with y components weren't being rotated properly. This has been fixed at a PR. However, this wasn't affecting the current criteria, as all rotated vectors only had *x* component.

PR #644 (2020-09-30): Rotate point bug
### Description

Fixed a bug at *rotate_point* function inside RunningRedLightTest criteria causing vectors to not be rotated properly. This doesn't change the criteria, as all the vectors rotated had no *y* component.

Fixes #643

#### Where has this been tested?

  * **Platform(s):** Ubutu 18.04
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.24
  * **CARLA version:** 0.9.10

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/644)
closed 2020-09-30 by fabianoboril