carla-simulator/scenario_runner #653: Scenario Runner via OpenSCENARIO waypoint mission for adversarial vehicles

SchDevel, 2020-10-05:
Hello together,

I am currently trying to get a simple scenario in the OpenSCENARIO format working.
In a first stage my goal is to have the evo_vehicle and one or multiple adversarial vehicle each on separate waypoint missions.

It is kind of related to the issue here: https://github.com/carla-simulator/scenario_runner/issues/516

However, the issue I linked above is with an old version of the scenario runner.

In my implementation ([WaypointScenario.zip](https://github.com/carla-simulator/scenario_runner/files/5328862/WaypointScenario.zip)), the one adversarial vehicle is not starting its route, even though I implemented a relative distance based trigger.

Do you have any idea, why this is the case?
Is having multiple adversarial vehicles on different waypoint missions even possible?

Best regards

SchDevel

 - OS: Ubuntu 20.04
 - CARLA Version 0.9.10
 - Python version 2.7
 - Version 0.9.10 (latest pull)

sagar-g-v, 2020-10-06:
Hi, I checked your file.

`adversary` is not moving because speed is set to `0.0` by default.

You are using only one event for `adversary` actor.

```xml
<Event name="FollowsRoute" priority="overwrite">
```

but in XOSC you need to use a seperated event to set the `adversary` Speed and then trigger `FollowsRoute` event.
- Exmple:
```xml
<Event name="adversarySetSpeed" priority="overwrite">
	<Action name="adversarySetSpeed">
		<PrivateAction>
			<LongitudinalAction>
				<SpeedAction>
					<SpeedActionDynamics dynamicsShape="step" value="10" dynamicsDimension="distance" />
					<SpeedActionTarget>
						<AbsoluteTargetSpeed value="30" />
					</SpeedActionTarget>
				</SpeedAction>
			</LongitudinalAction>
		</PrivateAction>
	</Action>
	<StartTrigger>
		<ConditionGroup>
			<Condition name="adversaryStartCondition" delay="0" conditionEdge="rising">
				<ByEntityCondition>
					<TriggeringEntities triggeringEntitiesRule="any">
						<EntityRef entityRef="ego_vehicle" />
					</TriggeringEntities>
					<EntityCondition>
						<RelativeDistanceCondition entityRef="adversary" relativeDistanceType="cartesianDistance" value="300" freespace="false" rule="lessThan" />
					</EntityCondition>
				</ByEntityCondition>
			</Condition>
		</ConditionGroup>
	</StartTrigger>
</Event>
<Event name="FollowsRoute" priority="overwrite">
	<Action name="FollowsRoute">
		<PrivateAction>
			<RoutingAction>
				<AssignRouteAction>
					<Route name="StraightRoute" closed="false">
						<ParameterDeclarations />
						<Waypoint routeStrategy="fastest">
							<Position>
								<WorldPosition x="-78.0" y="12.96" z="0.6" h="3.14" />
							</Position>
						</Waypoint>
						<Waypoint routeStrategy="fastest">
							<Position>
								<WorldPosition x="-11.42" y="43.82" z="0.6" h="1.57" />
							</Position>
						</Waypoint>
					</Route>
				</AssignRouteAction>
			</RoutingAction>
		</PrivateAction>
	</Action>
	<StartTrigger>
		<ConditionGroup>
			<Condition name="FollowsRouteStartCondition" delay="0" conditionEdge="rising">
				<ByEntityCondition>
					<TriggeringEntities triggeringEntitiesRule="any">
						<EntityRef entityRef="adversary" />
					</TriggeringEntities>
					<EntityCondition>
						<TraveledDistanceCondition value="10" />
					</EntityCondition>
				</ByEntityCondition>
			</Condition>
		</ConditionGroup>
	</StartTrigger>
</Event>

```

can you try this and let me know if it solves your issue.

SchDevel, 2020-10-06:
Hey @sagar-g-v,

thank you for the fast response!
Interesting point regarding the `adversary`speed and the trigger events. I did not know that.

However, adding this into the code did not change anything. The `adversary` vehicle is still not moving, despite being `<300` in cartesian coordinates away from the ego_vehicle.

I attached the file again here: 
[WaypointScenarioEdit.zip](https://github.com/carla-simulator/scenario_runner/files/5331862/WaypointScenarioEdit.zip)

Is there something wrong with the start triggers of the ManeuverGroup?

Thank you once again!

Best regards

SchDevel

seowwj, 2020-10-08:
Hi @SchDevel ,

I believe the issue is due to the start condition that has been set for the adversary.

Line 114: `<RelativeDistanceCondition entityRef="adversary" relativeDistanceType="cartesianDistance" value="300" freespace="false" rule="lessThan" />`

As the cartesian distance between ego and adversary is < 300 since spawning, hence the trigger condition does not run.
Can you try to set a shorter distance? Maybe having the condition at < 90?

For more info, you can refer ASAM's documentation [here](https://www.asam.net/index.php?eID=dumpFile&t=f&f=3496&token=df4fdaf41a8463e585495001cc3db3298b57d426#_conditions_and_triggers)
Below is an example of the trigger.
[image]

SchDevel, 2020-11-11:
Hey there,

sorry for my super late response. Thank you for your helpful comment @seowwj.
Playing with the start conditions solved my problem.
The reference to the documentation is very helpful.

Best regards
SchDevel

closed 2020-11-11 by SchDevel

EnekoRamirez, 2021-07-19:
> Hey there,
> 
> sorry for my super late response. Thank you for your helpful comment @seowwj.
> Playing with the start conditions solved my problem.
> The reference to the documentation is very helpful.
> 
> Best regards
> SchDevel

Hello! I am trying to run your .xosc code but it does not work. I spawn an ego vehicle in the simulation and afterwards execute your code. A second vehicle spawns behind the first ego-vehicle but none of them move. What conditions have you been playing with in order to move both vehicles?

Thank you 
Best regards

EnekoRamirez, 2021-07-19:
> Hello! I am trying to run your .xosc code but it does not work. I spawn an ego vehicle in the simulation and afterwards execute your code. A second vehicle spawns behind the first ego-vehicle but [...]

@SchDevel

ronyshaji, 2023-08-07:
@EnekoRamirez Did you found any solution to this issue?