carla-simulator/scenario_runner #654: Lane change action causing problems with triggers after it (OpenSCENARIO)

Lu7913, 2020-10-07:
Hi,
I'm creating simple overtaking scenario with OpenSCENARIO. In the scenario I perform lane change action and after this I would like to perform some other actions. The problem is that after lane change action no trigger is activated. I have tried different conditions and nothing works after lane change (Vehicle changes lane and drive on, another action is not triggered, no error is returned).
If I use those conditions with another action (like speed action, ...), new action are triggered normally. I was only using lane change attributes used also in demo.

I'm not sure if there is some kind of bug or I am missing something? I've been stuck here for few days now.

Thank you for you help.    

 - OS: Kubuntu 18.04
 - CARLA Version 9.10.1
 - ScenarioRunner Version 9.10

fabianoboril, 2020-10-08:
Hi,
can you share your scenariofile with us? This would help us to debug the issue.

Lu7913, 2020-10-08:
Did some more testing, so I performed two bare bones stopping the car scenarios. So the car is driving for some time, speed is changed and another speed action is triggered to stop the car, when first speed action is finished (StoryboardElementStateCondition). The car stops...

Second scenario is similar except instead of first speed change action, lane change is performed, and when it's finished, another speed change should stop the car, but it is not triggered (StoryboardElementStateCondition) and the car would not stop. 
Similar thing is with the ''RelativeDistanceCondition'' trigger. 

When designing similar scenario with only time condition trigger, the scenario is simulated as expected.
Did some digging in parser, if I would found something, but my knowledge of python is not sufficient.
Sorry for no file.

Lu7913, 2020-10-12:
Sorry for late response. 

```xml
<?xml version='1.0' encoding='UTF-8'?>
<OpenSCENARIO>
  <FileHeader author="" date="2020-03-24" description="CARLA:LaneChangeTest" revMajor="1" revMinor="0"/>
  <ParameterDeclarations/>
  <CatalogLocations/>
  <RoadNetwork>
    <LogicFile filepath="Town04"/>
    <SceneGraphFile filepath=""/>
  </RoadNetwork>
  <Entities>
    <ScenarioObject name="hero">
      <Vehicle name="vehicle.tesla.model3" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="ego_vehicle"/>
        </Properties>
      </Vehicle>
      <ObjectController/>
    </ScenarioObject>
    <ScenarioObject name="actor">
      <Vehicle name="vehicle.lincoln.mkz2017" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="simulation"/>
        </Properties>
      </Vehicle>
      <ObjectController/>
    </ScenarioObject>
  </Entities>
  <Storyboard>
    <Init>
      <Actions>
        <GlobalAction>
          <EnvironmentAction>
            <Environment name="Environment 1">
              <TimeOfDay animation="false" dateTime="2020-03-24T12:00:00"/>
              <Weather cloudState="skyOff">
                <Sun intensity="0.0" elevation="0.0" azimuth="0.0"/>
                <Fog visualRange="1000.0">
                  <BoundingBox>
                    <Center x="0.0" y="0.0" z="5.0"/>
                    <Dimensions width="10.0" length="10.0" height="10.0"/>
                  </BoundingBox>
                </Fog>
                <Precipitation precipitationType="dry" intensity="0.0"/>
              </Weather>
              <RoadCondition frictionScaleFactor="1.0"/>
            </Environment>
          </EnvironmentAction>
[... 110 lines cut ...]
              <ByEntityCondition>
                <TriggeringEntities triggeringEntitiesRule="any">
                  <EntityRef entityRef="actor"/>
                </TriggeringEntities>
                <EntityCondition>
                  <RelativeDistanceCondition freespace="false" rule="lessThan" entityRef="hero" value="17.0" relativeDistanceType="cartesianDistance"/>
                </EntityCondition>
              </ByEntityCondition>
            </Condition>
          </ConditionGroup>
        </StartTrigger>
      </Act>
    </Story>
    <StopTrigger>
      <ConditionGroup>
        <Condition delay="0.0" conditionEdge="rising" name="Time limit">
          <ByValueCondition>
            <SimulationTimeCondition rule="greaterThan" value="30.0"/>
          </ByValueCondition>
        </Condition>
      </ConditionGroup>
    </StopTrigger>
  </Storyboard>
</OpenSCENARIO>
```

Lu7913, 2020-10-19:
Maybe somebody tried to run scenario above? I'm still stuck here.

closed 2021-09-15 by Lu7913

pjk19942, 2023-10-03:
Did you find any solution to this?

Liuzy0908, 2023-10-20:
Same error. Did you find any solution to this?

Liuzy0908, 2023-10-20:
> Did you find any solution to this?

Same error. Did you find any solution to this?  Thanks
@pjk19942