carla-simulator/scenario_runner #655: Test & Support ALKS OpenSCENARIO examples

fabianoboril, 2020-10-08:
Some additional OSC examples were published here:
https://github.com/arauschert/OSC-ALKS-scenarios

Test and find out what we need to support.

arauschert, 2021-04-09:
Hi,

great that you already created an issue for support of the OSC-ALKS-scenarios!

I'm also trying to run them with CARLA now and here are my additional findings:
- error on loading openDRIVE map. see [here](https://forum.carla.org/t/error-on-loading-custom-opendrive-map/917)
- support for pedestrian catalog needed
- 4.2_3: the offset direction is inverted. the positive t axis of the road coordinate system goes to the left 
- 4.2_3: if Orientation is given in LanePosition, then the offset attribute is applied after the rotation and the entity coordinate system is used for the translation instead of the road coordinate system (pedestrian is turned 90° to the left and then moved 5 meters along the entity x coordinate which then equals to the road's s coordinate instead of the road's t coordinate)
- 4.3_1: The speed action doesn't do anything. DOn't know if it is not triggered (SimulationTimeCondition should work) or one of the attributes cannot be interpreted (maybe RelativeTargetSpeed?)

fabianoboril, 2021-04-15:
Hi Andreas,

thanks for your input. Let us try to get the things hunted down.

Regarding the debug out: Do you know about the --debug option? This will also provide you the complete behavior tree while the scenario is executing to get an understanding which part is not working as desired.

fabianoboril, 2021-04-15:
> Hi,
>
> great that you already created an issue for support of the OSC-ALKS-scenarios!
>
> I'm also trying to run them with CARLA now and here are my additional findings:
>
> * error on loading OpenDRIVE [...]

Could you please elaborate on the pedestrian catalog support. We have this supported and I tested your catalog and it seems to work just fine.

I also just updated the loading of the OpenDRIVE map, to avoid that the map is reloaded on every call (PR #752)

arauschert, 2021-04-15:
> Could you please elaborate on the pedestrian catalog support. We have this supported and I tested your catalog and it seems to work just fine.

Sure, thanks for investigating. Here's the output I get when running the scenario 4.2_1, where a pedestrian from the catalog is used as blocking target (actually I didn't see a difference with or without --debug option):

q496134@cmucl872167:~/CARLA/scenario_runner-0.9.11$ python scenario_runner.py --openscenario ../../OSC-ALKS-scenarios/Scenarios/ALKS_Scenario_4.2_1_FullyBlockingTarget_TEMPLATE.xosc --debug
WARNING:[SR:OpenScenarioConfiguration]: Wrong OpenDRIVE map in use. Forcing reload of CARLA world
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: cannot parse georeference: ''. Using default values. 
Preparing scenario: ALKS Scenario 4.2_1 FullyBlockingTarget Template
WARNING: Actor model car_ego not available. Using instead vehicle.tesla.model3
WARNING: Actor model Adult not available. Using instead walker.pedestrian.0001
The scenario cannot be loaded
Traceback (most recent call last):
  File "scenario_runner.py", line 374, in _load_and_run_scenario
    scenario = OpenScenario(world=self.world,
  File "/home/q496134/CARLA/scenario_runner-0.9.11/srunner/scenarios/open_scenario.py", line 187, in __init__
    super(OpenScenario, self).__init__("OpenScenario", ego_vehicles=ego_vehicles, config=config,
  File "/home/q496134/CARLA/scenario_runner-0.9.11/srunner/scenarios/basic_scenario.py", line 53, in __init__
    self._initialize_actors(config)
  File "/home/q496134/CARLA/scenario_runner-0.9.11/srunner/scenarios/basic_scenario.py", line 116, in _initialize_actors
    raise Exception("Error: Unable to add actors")
Exception: Error: Unable to add actors
Error: Unable to add actors
Destroying ego vehicle 87
ERROR: failed to destroy actor 87 : unable to destroy actor: not found 
No more scenarios .... Exiting

fabianoboril, 2021-04-16:
Which commit hash are you testing with?

arauschert, 2021-04-16:
I'm using release 0.9.11

fabianoboril, 2021-04-16:
Could you try testing #713. It is not yet completed but should resolve some of the issues you might see.

arauschert, 2021-04-19:
> Could you try testing #713? It is not yet completed but should resolve some of the issues you might see.

Hi, I checked out the branch and now the pedestrian scenarios run. Great! Seems these things are already fixed. 
Then let's concentrate on the other stuff.

arauschert, 2021-04-19:
Here's the output for scenario 4.3_1(tested with branch feat/OscControllerUpdates):

q496134@cmucl872167:~/CARLA/scenario_runner$ ./scenario_runner.py --openscenario ../../OSC-ALKS-scenarios/Scenarios/ALKS_Scenario_4.3_1_FollowLeadVehicleComfortable_TEMPLATE.xosc 
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: cannot parse georeference: ''. Using default values. 
Preparing scenario: ALKS Scenario 4.3_1 FollowLeadVehicleComfortable Template
WARNING: Actor model car_ego not available. Using instead vehicle.tesla.model3
WARNING: Actor model car not available. Using instead vehicle.tesla.model3
The scenario cannot be loaded
Traceback (most recent call last):
  File "./scenario_runner.py", line 375, in _load_and_run_scenario
    scenario = OpenScenario(world=self.world,
  File "/home/q496134/CARLA/scenario_runner/srunner/scenarios/open_scenario.py", line 187, in __init__
    super(OpenScenario, self).__init__(self.config.name, ego_vehicles=ego_vehicles, config=config,
  File "/home/q496134/CARLA/scenario_runner/srunner/scenarios/basic_scenario.py", line 63, in __init__
    behavior = self._create_behavior()
  File "/home/q496134/CARLA/scenario_runner/srunner/scenarios/open_scenario.py", line 325, in _create_behavior
    maneuver_behavior = OpenScenarioParser.convert_maneuver_to_atomic(
  File "/home/q496134/CARLA/scenario_runner/srunner/tools/openscenario_parser.py", line 1039, in convert_maneuver_to_atomic
    if 'role_name' in traffic_actor.attributes and traffic_actor.attributes['role_name'] == obj:
AttributeError: 'NoneType' object has no attribute 'attributes'
'NoneType' object has no attribute 'attributes'
Destroying ego vehicle 124
ERROR: failed to destroy actor 124 : unable to destroy actor: not found 
No more scenarios .... Exiting

fabianoboril, 2021-04-20:
A fix is integrated into the branch of #713.

arauschert, 2021-04-21:
> A fix is integrated into the branch of #713.

Yes, looks good. Now the scenario 4.3_1 starts. I'll mark this issue as done in the comment above.

arauschert, 2021-04-29:
The issue with the ego position at the start of the road (s=0) is now fixed in the newest version of the ALKS scenarios. They were also moved to the ASAM organization GitHub account:
https://github.com/asam-oss/OSC-ALKS-scenarios

arauschert, 2021-05-05:
One of the next topics could be the inversion of the direction of the lateral axis for positions in teleport actions according to the coordinate system definitions of OpenSCENARIO:
https://www.asam.net/index.php?eID=dumpFile&t=f&f=4092&token=d3b6a55e911b22179e3c0895fe2caae8f5492467#_coordinate_systems
I assume that then also other existing example scenarios in CARLA need to be adjusted?

ralwing, 2022-04-27:
I tried to test those scenarios tag/0.3.2 in Carla 0.9.13 + Scenario Runner master with merged #713 without luck:

```
carla@GiGis:~/scenario_runner$ python3 scenario_runner.py --openscenario /home/carla/OSC-ALKS-scenarios/Scenarios/ALKS_Scenario_4.2_1_FullyBlockingTarget_TEMPLATE.xosc 
WARNING:[SR:OpenScenarioConfiguration]: Wrong OpenDRIVE map in use. Forcing reload of CARLA world
WARNING: cannot parse georeference: ''. Using default values. 
Traceback (most recent call last):
  File "scenario_runner.py", line 607, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 507, in run
    result = self._run_openscenario()
  File "scenario_runner.py", line 495, in _run_openscenario
    config = OpenScenarioConfiguration(self._args.openscenario, self.client, openscenario_params)
  File "/home/carla/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 61, in __init__
    self._parse_openscenario_configuration()
  File "/home/carla/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 93, in _parse_openscenario_configuration
    self._set_actor_information()
  File "/home/carla/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 251, in _set_actor_information
    entry = OpenScenarioParser.get_catalog_entry(self.catalogs, catalog_reference)
  File "/home/carla/scenario_runner/srunner/tools/openscenario_parser.py", line 365, in get_catalog_entry
    entry = catalogs[catalog_reference.attrib.get("catalogName")][entry_name]
KeyError: '$TargetBlocking_Catalog'
carla@GiGis:~/scenario_runner$ 
```
I've also tried the current master, still no luck:
```
carla@GiGis:~/scenario_runner$ python3 scenario_runner.py --openscenario /home/carla/OSC-ALKS-scenarios/Scenarios/ALKS_Scenario_4.2_1_FullyBlockingTarget_TEMPLATE.xosc 
Traceback (most recent call last):
  File "scenario_runner.py", line 607, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 507, in run
    result = self._run_openscenario()
  File "scenario_runner.py", line 495, in _run_openscenario
    config = OpenScenarioConfiguration(self._args.openscenario, self.client, openscenario_params)
  File "/home/carla/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 61, in __init__
    self._parse_openscenario_configuration()
  File "/home/carla/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 89, in _parse_openscenario_configuration
    self._check_version()
  File "/home/carla/scenario_runner/srunner/scenarioconfigs/openscenario_configuration.py", line 103, in _check_version
    raise AttributeError("Only OpenSCENARIO 1.0 is supported")
AttributeError: Only OpenSCENARIO 1.0 is supported

```
what's intended... 

What changes do i need to make it work?

arauschert, 2022-04-28:
Hi! Great that you try to make CARLA compatible to OpenSCENARIO!
The upper error looks like the runner couldn't resolve the catalog name $TargetBlocking_Catalog. The parameter is defined in the parameter declaration section with the value "PedestrianCatalog". Was the parameter replaced by it's value, so the runner looks for the key "PedestrianCatalog" and not for "$TargetBlocking_Catalog"?

ralwing, 2022-04-28:
Thank You @arauschert. A one small change for me, one giant leap for mankind ;-)

christinaSak, 2022-05-19:
Hey @arauschert, I have the same issue when I try to run scenario 4.3_2 as you mentioned above: "4.3_2: The speed action doesn't do anything. Don't know if it is not triggered (SimulationTimeCondition should work) or one of the attributes cannot be interpreted". Did you manage to make any progress?

arauschert, 2022-05-19:
> Hi @arauschert, I have the same issue when I try to run scenario 4.3_2 as you mentioned above: "4.3_2: The speed action doesn't do anything. Don't know if it is not triggered (SimulationTimeCondition should work) or one of the attributes cannot be interpreted". Did you manage to make any progress?

Hi! Probably the acceleration/deceleration feature is contained in the branch of this PR https://github.com/carla-simulator/scenario_runner/pull/713 . Can you please try with this scenario_runner version?

christinaSak, 2022-05-19:
> > Hi @arauschert, I have the same issue when I try to run scenario 4.3_2 as you mentioned above: "4.3_2: The speed action doesn't do anything. Don't know if it is not triggered (SimulationTimeCondition should work) or one of the attributes cannot be interpreted". Did you manage to make any progress?
> 
> Hi @christinaSak ! Probably the acceleration/deceleration feature is contained in the branch of this PR #713 . Can you please try with this scenario_runner version?

Thank you now it works fine, probably the AbsoluteTargetSpeed in SpeedActionTarget was not supported in the version I was using.

rupeshreddy97, 2022-11-08:
Hi @arauschert. I have been running the ALKS scenarios with Carla and scenario runner but during the simulation I don't think the ALKS feature is being implemented in the scenario. For example in ALKS_Scenario_4.2_1_FullyBlockingTarget_TEMPLATE, the vehicle doesn't stop for the Pedestrian. I tried in multiple versions of Carla but that didn't help. Is there something I'm missing regarding the Controller part? Please suggest!

rupeshreddy97, 2022-11-09:
Thank you @arauschert for your input! This would really help us moving forward.

fatemehprhm, 2023-08-01:
Hi, Can you please add support for SpeedProfileAction? It really makes scenarios more simple to generate.

Kai-tang-Hu, 2023-09-19:
> Hi! Great that you try to make CARLA compatible to OpenSCENARIO! The upper error looks like the runner couldn't find a catalog with the given name. The parameter is defined in the parameter declaration section with the value "PedestrianCatalog". Was the parameter reference replaced by it's value, so the runner looks for the key "PedestrianCatalog" and not for "$TargetBlocking_Catalog"?

I have the same issue. I don't where I should change to make the runner find correct Catalog. Could you help me with that?

Kai-tang-Hu, 2023-09-20:
> Thank You @arauschert. A one small change for me, one giant leap for mankind ;-)

Did you get the code to work? What did you change for this code?