carla-simulator/scenario_runner #660: Deploying custom agents on custom scenarios

aaronh65, 2020-10-13:
I'd like to deploy a custom agent (something like the Learning by Cheating approach) on custom scenarios, but it looks like this is only supported through RouteScenarios. I haven't been able to find a tutorial on how to set up RouteScenarios of my own creation, is there a recommended way of approaching this?

On another note, RouteScenarios seem quite heavyweight for my use case - I'm interested in individual scenarios e.g. a specific intersection, and I'd like to have control over things like traffic light state, position, or perhaps replacing traffic lights with stop signs. Are there plans to support non-RouteScenario scenarios for deployment with custom agents?

glopezdiest, 2020-10-14:
Hey @aaronh65. All agents are indeed supported for scenarios outside routes, it is just that in the case of routes, they are run by SR directly, so they need to be given. However, the same way you can run the manual control on a separate terminal, you can create your own agent or modify the owns available at the PythonAPI (*carla/PythonAPI/carla/agents/navigation*), and they'll work just fine. Also, take into account that they will have no route given to them, so it might be trickier (or maybe not).

> RouteScenarios seem quite heavyweight for my use case

I have to agree that routes seem very daunting at first, but it's just that there are no easy examples given, and the first thing you see are very long routes with tons of scenarios plus some background activity in the middle. However, you can reduce the amount of scenarios, remove the background activity and makes routes last for only some hundred meters, and they'll seem much simpler. You can actually just make the route long enough to have your desired scenario, and that's it.

> I'd like to have control over things like traffic light state, position, or perhaps replacing traffic lights with stop signs.

You can do most of that via the atomics SR provides. I don't think you can move / remove traffic lights or spawn stop signs though during run-time, you'd have to use the Unreal Editor instead of the package, and change them there before starting the simulation (or create a new map altogether)

wnklmx, 2021-03-18:
Hi @aaronh65 and @glopezdiest!
I'm right now working on a similar 'problem' since I want to test agents (in my case the [pylot ](https://github.com/erdos-project/pylot)agent) in single scenarios.

What I've done so far is pretty much what @glopezdiest suggested:
> However, you can reduce the amount of scenarios, remove the background activity and makes routes last for only some hundred meters, and they'll seem much simpler. You can actually just make the route long enough to have your desired scenario, and that's it.

I pushed my changes to my [fork](https://github.com/wnklmx/scenario_runner/tree/master) so you might have a look.
If there is interest I could do some further testing and open pull requests for these changes.

closed 2021-11-23 by glopezdiest