carla-simulator/scenario_runner #663: question about stable version

sagar-g-v, 2020-10-15:
Hi,

I have been using scenario runner 0.9.9 till now.

now i want to start using 0.9.10 and its new features.

But 0.9.10 is still not stable to satart using it completely.

can i know how long will it take to make current version as stable?

if any help needed to evaluate the functionalties. I can help.

Regards,
Sagar G V

fabianoboril, 2020-10-15:
Hi,
do you refer to that there is no 0.9.10 branch? You could use the tag v0.9.10, or simply stick to master. API breaking changes should not happen to many now, since we mostly work on bringing in new OSC features, which should happen under the hood.

sagar-g-v, 2020-10-15:
I think sticking to `master` makes sense.
I will evaluate master branch and find issue and fixes.

Thank you.

closed 2020-10-15 by sagar-g-v